hzeller/rpi-rgb-led-matrix #598: Unused GPIO Pins

mblue002, 2018-04-22:
Hello!
I have a chain of 3 32x32 panels working now. Currently hand-wired, as the Adafruit Hat I was using had some issues. With my chinese P6 (3535) 1/8 scanning panels, I need these additional settings:
--led-multiplexing=6 --led-pwm-bits=1 --led-shadow-gpio=3

My application only requires the 96x32 display on a single chain. But I have a need to use some GPIO for other control functions. Is it possible to use the GPIO pins assigned to the second and third chains by limiting their use in the rpi-rgb-led-matrix? For instance, I would like to use pins 29,31,32,33,35,36,37,38,40 (assigned for [2]=💥 and [3]=💧) as inputs.

I took a look at gpio.h and gpio.cc but its not clear to me how to go about this. Is it even possible?

Thanks in advance.

mblue002, 2018-04-22:
After further review, it seems that Framebuffer::InitGPIO() only uses the gpio pins that are required according to the setting of --led-parallel. Since my setting and the default is --led-parallel=1, the unused gpio pins appear to be available. I have tested this with a quick test by connecting PIN37 and PIN38 together, then used a quick python to monitor PIN38 while setting PIN37 HIGH and LOW. This seems to work while simultaneously displaying an image on the 3 panels.

So, I will wait a few days for someone to respond to this issue and tell me that I should not be taking this approach, or, hopefully verifying that this IS an acceptable approach. 

Thanks again.

hzeller, 2018-04-27:
Yes, you can use the remaining GPIOs in your code.