hzeller/rpi-rgb-led-matrix #599: Unable to bring the LED matrix to turn

lazuhrow93, 2018-04-25:
Hello all,

I am using python. 

So I believe I have all the wiring correct but I can't seem to get it up and running. I've tried to get the LED to work with this tutorial: https://learn.adafruit.com/connecting-a-16x32-rgb-led-matrix-panel-to-a-raspberry-pi/experimental-python-code and it turns on as expected. I am able to get it wokring with that tutorial so im assuming my wiring is good. I also have a 64x32 LED matrix. I eventually want to chain another 64x32. (128x32)

My inspiration for my project uses this rpi-rgb-led-matrix so I came here to try it out. The steps I took:
1. Went to root directory of the cloned repository.
2. entered the commands to build python 2 (I used python -V to check which version i am using. 2.7.13)
3. CD to python/samples
4. then entered: sudo ./pulsing-colors.py --led-rows=32 --led-cols=64 --led-no-hardware-pulse=True

In the terminal the program looks like its working

Am I missing something? Ive also included pics of the LED matrix just in case that helps. I am still looking to see if I am missing something myself. Your help is much appreciated
[image]
[image]
[image]

hzeller, 2018-04-25:
You need to follow the wiring described in this project: https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md

lazuhrow93, 2018-04-25:
thanks, that helped.

closed 2018-04-25 by lazuhrow93