hzeller/rpi-rgb-led-matrix #600: Problem with Scrolling Text with  pixelmapper:rotate:90

pabaycroft, 2018-04-27:
When using the pixelmapper with rotate values of 90 or 270, scrolling text has a 'ripple' effect on the width which makes it difficult to read. Rotate values of 0 or 180 don't have this problem. The ripple becomes more pronounced the faster the text scrolls. Static displays are fine.
To reproduce this, use the runtext.py example with a few words as the -t value and with pixelmapper:rotate=0, change the time.sleep value to 0.025 to increase the scrolling speed, and note the smooth scrolling. Now change to pixelmapper:rotate=90, run the program again to see the problem. Next change the time.sleep value to 0.01 to see the effect of a higher scrolling speed.
Any ideas what the cause may be ? I used a 'rotated' display for scrolling text without problems in early 2017 with an earlier version of the program using a command-line option (-R ?). Can supply more detail of my system if needed.
Many thanks,
Paul Baycroft

hzeller, 2018-04-27:
The rotate is probably not directly the issue, as it does not have any performance impact (internally, there is just a look-up table).

I suspect what is happening is, that the refresh of the display interferes with the scrolling in a way that makes it visually have that effect. Make sure to use `SwapOnVSync()`' and optimize your refresh rate with the usual means (e.g. `--led-pwm-dither-bits=1`)

pabaycroft, 2018-04-27:
Thanks for the very quick reply !
1. Yes am using SwapOnVSync
2. Couldn't find a way to use the  --led-pwm-dither-bits=1. It was rejected as an 'unrecognized parameter' by the Python interpreter, and couldn't see a way to use it in the matrix options which I use to set the chain length etc.
However, following the theme of a possible timing issue, I found that increasing the time for the pwm lsb from the default 130ns  to 300ns, made the problem worse !
What I've done as a 'fix' is to reduce the pwm-lsb-nanoseconds to the minimum of 50ns, increased the gpio_slowdown to 2, and applied these via the matrix.ptions only when the Rotate value is 90 or 270. The downside is a small amount of random speckling, but it's hardly noticeable.
Thanks again
Paul Baycroft

closed 2018-04-30 by pabaycroft