hzeller/rpi-rgb-led-matrix #601: ZStripe Inverted stripes

ctodobom, 2018-04-28:
Hi, I have some panel that are ZStripes multiplexed, but the stripes shows inverted.

I made this patch to fix it:

```patch
diff --git a/lib/multiplex-mappers.cc b/lib/multiplex-mappers.cc
index a48ddb9..a47eb14 100644
--- a/lib/multiplex-mappers.cc
+++ b/lib/multiplex-mappers.cc
@@ -191,6 +191,7 @@ static MuxMapperList *CreateMultiplexMapperList() {
   result->push_back(new CheckeredMultiplexMapper());
   result->push_back(new SpiralMultiplexMapper());
   result->push_back(new ZStripeMultiplexMapper("ZStripe", 0, 8));
+  result->push_back(new ZStripeMultiplexMapper("ZStripeUneven", 8, 0));
   result->push_back(new ZStripeMultiplexMapper("ZnMirrorZStripe", 4, 4));
   result->push_back(new CoremanMapper());
```

phonorad, 2018-04-30:
I need to make this same change for the panels I am using.  I am new with Python, though, so please bear with me.  Should the same thing happen to led-multiplexing=4 is I make the change (swap 0 and 8) and run make in the 'matrix' directory (seems to recompile multiplex mappers.cc), then run 'make build-python' and 'sudo make install-python'.  

I am running a python program that uses the rgbmatrix library, but the change does not seem to change the mapping for the python program (it is an mlb scoreboard program), although making the change as above does make the panel work properly with the demos.

The calling program only works up to led-multiplexing=4, so if I should be applying your patch above, how would I apply it (what are directories 'a' and 'b', do I make a file.py with your code in it and just run patch in the lib directory?), and does it add led-multiplexing option #'s or does it just replace option 4 (which is what I really need to do) with the reversed 8,0?

Or am I way off base all of this?

hzeller, 2018-04-30:
If you change the c++-code, make sure to recompile and re-install the Python binding.

hzeller, 2018-04-30:
... also, you need to make sure to be able to choose the number of the multiplex mapper.

I have now updated the code to include @ctodobom 's change and made the choice available in Python.

closed 2018-04-30 by hzeller

phonorad, 2018-04-30:
As far as recompiling, when I run make from the matrix library, it seems to recompile the multiplex-mappers.cc file, right?
Yes, I also ran 'sudo pip install -e python', which I think reinstalls the python binding.
I seem to be missing something, though, to make the mlb scoreboard program pick up the mapping change, as it does the same thing whether I have the zstripe settings at 0,8 or 8,0, while the demos (D=3, for example) definitely change the mapping from incorrect to correct, so at least enough of a recompile and rebuild to get the demos working seems to be occurring.

phonorad, 2018-04-30:
Sorry I missed your last comment while I was typing.  The author of the MLB program only updated it to take up to led-multiplexing=4, so I was trying to make that work.  If you have published the change, I can ask him to update again, I assume the ctodobom change must be option 8, but i will go check and then see if I can get him to update again.

hzeller, 2018-04-30:
For now, just remove the `choices=[0,1,2,3,4]` part in your flags of the program that limits the choices.

To [re-install the python part](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/bindings/python), you need to call
```
sudo make install-python
```

ajbowler, 2018-04-30:
@hzeller much appreciated, the program @phonorad was referring to was my [LED scoreboard for MLB](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard) which pulls this project in as a submodule.

I'll remove the choices so if I need to update this again I'll only need to get the submodule updated. Thanks a bunch.

commit 36a46331 (2018-04-30): Add ZStripUneven mapper (Fixes #601)