hzeller/rpi-rgb-led-matrix #603: Python --led-row-addr-type=2

ToXIcGaming, 2018-04-29:
I am able to get my panel (Chinese P10 32x16) to display properly using the `--led-rows=16 --led-row-addr-type=2 --led-multiplexing=3` with the demos but `--led-row-addr-type=2` is not available in the Python options?

hzeller, 2018-04-29:
The Python options struct has the row_addr_type field which you can set.

It also has the `--led-row-addr-type` command line flag if you use the [`samplebase.py`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/bindings/python/samples/samplebase.py#L28)

ToXIcGaming, 2018-04-29:
Ah yes, I fixed it by adding a option 2 to the `--led-row-addr-type` line in the `samplebase.py`

Specifically I changed this line  `self.parser.add_argument("--led-row-addr-type", action="store", help="0 = default; 1=AB-addressed panels", default=0, type=int, choices=[0,1])`

to  ` self.parser.add_argument("--led-row-addr-type", action="store", help="0 = default; 1=AB-addressed panels, 2 = direct row select", default=0, type=int, choices=[0,1,2])`

hzeller, 2018-04-29:
Thanks, now fixed in head.

commit ebefb916 (2018-04-29): Add missing row direct choice for row-addr type in Python. Fixes #603
closed 2018-04-29 by hzeller