hzeller/rpi-rgb-led-matrix #604: Single Passive RPi2 (40 Pin) Adapter?

Pghpunkid, 2018-05-01:
Not so much an issue, as it is a request but...

I see that there is a single passive adapter for older Pi's and older pinout method, but only a Passive 3 for 40 Pin RPi models.

Is there anyway we can get a single passive for 40 Pin and keep it as minimal as possible so we might be able to fit the PCB itself in a case (and cut the case for the HUB75 connector)? I tried to look at KiCad to modify the Passive 3 to do this so OshPark could make them, but basically immediately froze up when I saw the design.

hzeller, 2018-05-01:
Having long cables directly connected to the Pi is a little bit dangerous as it is easy to fry the GPIO pins (they are directly wired into the CPU without any safety in-between). Also, the LED matrixes often have issues with 3.3V logic levels.
So I suggest to just solder up an active adapter.
Alternatively: Somewhere in the issue list people were mentioning that there was a Chinese company that made ready-made active adapters that are 1:1 compatible with this library, that might also be an option (they were really cheap assembled, <$5 IIRC).

Pghpunkid, 2018-05-01:
And as far as the Chinese adapters, I have been searching everywhere for premade adapters, and haven't found much. If you know of a place, id love to take a look. However if there was a single active board at OSHPark that the board fit in a case (Upper clearance is okay, so long as I can cut holes the case and have it clasp together and screw together still).

maddogs, 2018-05-01:
The adaptors can be purchased here

http://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/

I'm using a couple and they work fine.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828