hzeller/rpi-rgb-led-matrix #606: Rotate 90 not working as expected with multiple panels

cmagnuson, 2018-05-01:
I'm running a chain of 5x 16x32 panels rotated 90° with this command:
`sudo rpi-rgb-led-matrix/examples-api-use/text-example --led-gpio-mapping=adafruit-hat --led-chain=5 --led-rows=16 --led-cols=32 --led-row-addr-type=2 --led-multiplexing=8 -f ter-powerline-x32n.bdf --led-pwm-bits=1 --led-pwm-lsb-nanoseconds=1000 --led-slowdown-gpio=2 --led-pixel-mapper="Rotate:90"`

But entering a string only results in the first character being printed.  I can write one character in each panel by typing:

> T\nE\nS\nT\n

 for example, but I'd expect to be able to write out that string using just:

> TEST\n

I'd guess it is still assuming the same canvas width as before the rotation?

The scrolling-text-example has similar results - it only displays on the first panel when "Rotate:90" is set.  When I remove the pixel mapper though it will draw all panels as expected.

Scrolling text example:
[image]

hzeller, 2018-05-02:
(you have not given a video of the scrolling example to illustrate).

If you have a chain of 5 and rotate it by 90 degrees, you have a portrait display, so now it is very narrow to display the texts, so it sounds like from what you describe like an expected result.

If you want to have text that writes a vertical text from top to bottom on such a screen, you might want to use `VerticalDrawText(...)`

cmagnuson, 2018-05-02:
Thanks for the explanation, I just misunderstood how the rotate command was working.

In the initial case (no rotate) I have a long horizontal row of panels (case 1).  After setting rotate 90° I'd expected to have a horizontal row of panels with each individual panel rotated 90° (case 2) but as you described I see it takes the entire matrix of panels and rotates that by 90° (case 3).

[image]

So I was incorrect in my assumption and the rotate is working.  Is there a way to apply a rotation to the individual panels instead of the entire matrix?

angelogoncalve, 2018-05-02:
see this issue https://github.com/hzeller/rpi-rgb-led-matrix/issues/368

angelogoncalve, 2018-05-02:
See rhis issues also:

https://github.com/hzeller/rpi-rgb-led-matrix/issues/552

https://github.com/hzeller/rpi-rgb-led-matrix/issues/481

cmagnuson, 2018-05-03:
Thanks for the pointers.

Would you consider a pull request for a PixelMapper like this:
```
class RotateSinglePanelPixelMapper : public PixelMapper {
public:
  RotateSinglePanelPixelMapper() : chain_(0) {}

  virtual const char *GetName() const { return "RotatePanel"; }

  virtual bool SetParameters(int chain, int parallel, const char *param) {
    chain_ = chain;
    return true;
  }

  virtual bool GetSizeMapping(int matrix_width, int matrix_height,
                              int *visible_width, int *visible_height)
    const {
      *visible_width = matrix_height * chain_;
      *visible_height = matrix_width / chain_;
    return true;
  }

  virtual void MapVisibleToMatrix(int matrix_width, int matrix_height,
                                  int x, int y,
                                  int *matrix_x, int *matrix_y) const {

  const int panel_width = matrix_width / chain_;
  const int panel_height = matrix_height;
    
  *matrix_x = matrix_width - (y + panel_width * (x / panel_height)) - 1;
  *matrix_y = x % panel_height;
}

private:
  int chain_;
};
```

It is working for the use case I described above, but doesn't handle parallel panels at the moment.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828

closed 2025-11-27 by marcmerlin