hzeller/rpi-rgb-led-matrix #608: Python Image-viewer.py Green and Blue Swapped

henrylowzh, 2018-05-02:
Greetings @hzeller ,

Have anyone came across this issue whereby the blue and green colours are swapped when previewing image on the python image-viewer.py? Is there a way to reverse this?

Original Image:
[image]

Output Image on LED panel:
[image]

```
#!/usr/bin/env python
import time
import sys

from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image

if len(sys.argv) < 2:
    sys.exit("Require an image argument")
else:
    image_file = sys.argv[1]

image = Image.open(image_file)

# Configuration for the matrix
options = RGBMatrixOptions()
options.rows = 64
options.cols = 128
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat-pwm'  # If you have an Adafruit HAT: 'adafruit-hat'

options.gpio_slowdown = 4
options.pwm_lsb_nanoseconds =130
options.pwm_bits = 1

matrix = RGBMatrix(options = options)

# Make image fit our screen.
image.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)

matrix.SetImage(image.convert("RGB").rotate(180))

try:
    print("Press CTRL-C to stop.")
    while True:
        time.sleep(100)
except KeyboardInterrupt:
    sys.exit(0)
```

Best Regards,
Henry

kentrosaur, 2018-05-12:
Hi,

I had to change the defaults for the --led-rgb-sequence, try out the command line switches to test what layout you have. 

To make permanent for the python examples:  in samplebase.py, change the RGB line to RBG:
        self.parser.add_argument("--led-rgb-sequence", action="store", help="Switch if your matrix has led colors swapped. Default: RBG", default="RBG", type=str)

For the other programs, you can change the default in  lib/led-matrix.cc:  led_rgb_sequence("RBG"),

flavio-fernandes, 2018-09-19:
It seems to me that the char* used to store the led_sequence is pointing to memory that has been feed up. I only see this issue if I use python and set options to something like:
options.led_rgb_sequence = "RBG"
```
$ sudo ./screen.py
LED sequence '
�t' does not contain any 'R'.
Aborted
```
@kentrosaur : any chance the binding should be strdup() like in **led_matrix_create_from_options** ?

smallwat3r, 2018-12-04:
I don't think the colours are swapped. I think you just have to remove `.rotate(180)` when you set the matrix image ie. `matrix.SetImage(image.convert("RGB").rotate(180))` to just `matrix.SetImage(image.convert("RGB"))`.