hzeller/rpi-rgb-led-matrix #613: Raspberry Pi 3 64 bit support

vifino, 2018-05-06:
Hi!
There is a Gentoo 64 bit image floating around [here](https://github.com/sakaki-/gentoo-on-rpi3-64bit).
The library does not seem to work when run there. There is simply no error message or output of any kind.

I've booted the kernel with `iomem=relaxed` (i think).

Perhaps you can look into this to get an additional bit of performance on the Pi 3?

vifino, 2018-05-17:
I'll try that in a few hours, will report back.

vifino, 2018-05-18:
Sorry for the long wait, the Pi 3 I intended to test it with is lost somewhere. Got the alternative and set it up, testing it with the Pi 2 detection routine patched revealed that, unfortunately, it's not that:

See http://0x0.st/seJB.txt for the kernel log.

Hoping you can extract some use out of that.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828