hzeller/rpi-rgb-led-matrix #614: 16x32 HUB75 display only half of the board

paanjailani, 2018-05-08:
Hello ! Thanks for this project as reference. I have problem which is my LED panel just displayed half (the other half rows remain in light blue color)  . Can you assist me on this ? I will give some pictures of it

paanjailani, 2018-05-08:
This is sample of it

http://prntscr.com/jf5jvq

paanjailani, 2018-05-08:
[http://prntscr.com/jf5l3h](url)

paanjailani, 2018-05-08:
[image]

paanjailani, 2018-05-08:
[image]

hzeller, 2018-05-17:
This looks like a wiring issue or the panel is broken.
Given that all three colors of the lower half don't seem to work (R2, G2, B2), your panel is probably broken.

hzeller, 2018-05-21:
Make sure that you properly connect the grounds and have digital voltage converters between the Pi and the board (e.g. HCT245; I suggest to just solder up one of [the adapters](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3). The artifacts between the pixels look like you have floating voltages.
Also make sure you have short cables running to your matrix, the ones in your picture look very long.

It might help if you set the `--led-slowdown-gpio` higher, but you fundamentally have to fix your electrical issues.

paanjailani, 2018-05-21:
Thanks again. Will get the adapter ready and will get back to you once
again after it had been done.! Thanks again, really appreciate your help :)