hzeller/rpi-rgb-led-matrix #619: Trouble using ledcat

natcl, 2018-05-13:
Hello,
I can run the normal examples fine but I'm having troubles using the ledcat example, would it be possible to provide an example command ?
Thanks !

natcl, 2018-05-24:
Thanks that was it ! Works now !

closed 2018-05-24 by natcl

hzeller, 2018-05-24:
Hey, if any of you would like to write a little section in the [example-api-use/README.md]( https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/README.md) about how to use `ledcat`, that would probably help the next person tremendously. Can you send a pull request ?

boutell, 2020-10-01:
For the next person with an Adafruit hat or bonnet:

ledcat.cc overrides the HARDWARE_DESC setting you may have changed in `lib/Makefile` with this line...

```
defaults.hardware_mapping = "regular";
```

If you remove that line it should behave more like the other utilities. In particular if you set `HARDWARE_DESC?=adafruit-hat` in `lib/Makefile` before compiling then `ledcat` will work "out of the box" on your adafruit hat or bonnet just like the other utilities do. You will only have to pass `--led-rows` and `--led-cols`.

I think this line should be removed. `input_example` and `minimal_example` still have it too; the others do not.

I'll submit a PR.