hzeller/rpi-rgb-led-matrix #621: Port for Orange Pi H3

hubshi, 2018-05-15:
Dear!

Due to the fact that Raspi is lacking internal storage and SD Cards are not a viable option for long term and secure operation. I am using a Orange Pi Plus 2 which is equipped with internal EMMC and large RAM, while the famous WiringPi library is also ported for this platform, Is it possible to port this amazing piece of software to Orange Pi ? 

Best Regards.

maddogs, 2018-05-15:
You could always use a Raspberry Pi 3 and set it up to use USB to SSD instead of an SD Card.

hubshi, 2018-05-17:
Well, we use our embedded solutions in industry and ssd is not cheap in my area , while clients always want cheap and secure solutions ;) , Chinese products and pricing model made clients wants much for lesser price, so an internal EMMC sounds promising.

depili, 2018-05-17:
You can always netboot your Raspberry pi 3's. If you want a secure
solution the chinease banana/orange pi's aren't it, because the only
platform you can get for them is a chinease precompiled kernel with
unknown modifications. The kernels have already been shown to contain
backdoors.

https://securityaffairs.co/wordpress/47272/hacking/kernel-backdoor-allwinner.html

The chinease pi clones look really promising on paper but unfortunately
the reality is that they are next to useless. Software support is horrible
and they go through so many unnannounced hardware revisions that you
usually can't get the same product later...

hubshi, 2018-05-17:
I am using Armbian [](https://www.armbian.com/) which is i hope is not using a Chinese kernal. IMHO .

Anyway the original question still stands there , is it possible to port rgb-led-matrix to other platforms?

depili, 2018-05-17:
Even armbian needs to use the chinease kernel modifications, firmware and
bootloaders as even when the allwinner specifications are theoretically
open duplicating the chinease code would require major work per chip
variant. And since the life-time of a single allwinner chip variant is so
short why bother writing the needed modules...

hzeller, 2018-05-17:
The OrangePi is a completely different machine with a completely different GPIO subsystem, so you'd have to port the functionality of [gpio.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/gpio.cc) to it so that it can be used as a replacement.
But I don't have an OrangePi to test, so it is up to you to do the port and send a pull request.

WiringPi is not an option as we need high-speed memory-mapped parallel access to GPIO and ideally a high-speed timer that can directly output to a GPIO pin. The former you can probably do with the OrangePi, the latter you have to consult the datasheet of their processor.

hubshi, 2018-05-19:
Thank you Hzeller , it helps a lot, I will look into it and will try to work on it , thanks a lot 🥇

hubshi, 2018-05-19:
Thanks @depili for your insight, but now OrangePi is the only option ,  Thanks a lot 👍

forstermf, 2020-12-04:
@hubshi  how did the orang pi work out?

unic8s, 2025-02-03:
A little bit late to this party but have some of you every noticed this repo?
https://github.com/Edragon/rpi-rgb-led-matrix-rockchip

There are 2 ports (Allwinder and Rockchip) which seem to be still experimental, but have been successfully tested.
Demo video https://twitter.com/electro_phoenix/status/1522157916757041152

Here is something like a project page which also contains info for getting a driver hat for the OrangePis.

marcmerlin, 2025-02-03:
let's merge both issues

closed 2025-02-03 by marcmerlin