hzeller/rpi-rgb-led-matrix #624: 64x32 LED Matrix Wiring Issue

jlrasp, 2018-05-17:
I have a currently working 16x32 LED Matrix, bought a second 64x32. Wired per diagram with latest Raspberry 3 and not getting correct output. Here is example from running demo -D 9, I've tried with --led-chain=2, I've tried with --led-rows=32 --led-cols=64. I've tried matching the wiring exactly as the smaller screen, still same issue, hub75b connector on both from what I can tell. Any hints or help anyone can provide? I don't know what else to try.

Working: (16x32)
https://imgur.com/gKULkRZ

Non Working: (64x32)
https://imgur.com/LZruQfo

hzeller, 2018-05-18:
Have you tested the various `--led-row-addr-type` options ?

jlrasp, 2018-05-18:
Thank you for your feedback.
I've just tried these various options, they move pixels but its basically still the same.
sudo ./demo -D 9 --led-row-addr-type=0,1,2
0
https://imgur.com/bvJrE4b
1
https://imgur.com/Si0wnc5
2
https://imgur.com/P177Suf

It's probably a wiring issue but I thought reading through the wiring page, it pretty much would be the same wiring as my smaller led panel but no luck, I've done the rewiring a few times, from scratch, from my other panel.  Do  you think its possibly something with my panel?

jlrasp, 2018-05-18:
I've resolved the issue as it appears to be the Pi itself (Raspberry Pi 3 - Model B+ - 1.4GHz Cortex-A53 with 1GB RAM), when I hook it up to my other pi (which is a Pi3 but not the latest model), it works with my wiring. Assuming the wiring hasn't changed between Pi models. I've created this wiring diagram as well since I was having a hard time translating between hub75 and the pi wiring, if it helps anyone else, great.
My wiring:
https://imgur.com/QPGNjHv

Thanks again.

closed 2018-05-18 by jlrasp

sahukanishka, 2020-03-24:
the connection you show is not working in my case. I am also using 64x32 outdoor p4 panels with pi b+ model and everything is the same.
please help 
Thanks