hzeller/rpi-rgb-led-matrix #626: Displaying animated GIFs with Python

onionsTasteGood, 2018-05-24:
Unlike C++, when displaying an animated GIF with Python, the image displays but only the first frame. Does this library provide a solution for this or should I look to add this functionality to the code myself? If so, what would be the best way to do this? Through GraphicsMagick maybe? Any hints would be greatly appreciated.

I took a look at a [previous issue](https://github.com/hzeller/rpi-rgb-led-matrix/issues/11), but it's two years old, and I thought maybe there's a solution now.

Thanks!

hzeller, 2018-05-24:
From the library usage point of view, you essentially want to create an offscreen canvas, and then swap images and send one at a time. and use the off-time to fill the next image and swap. Have a look at the examples of how to use the SwapOnVSync().
If you only have a few images in the gif, it might be useful to preprocess all of them into their own offscreen canvas and in the main loop then just swap these in one at a time - the CPUs are often overwhelmed to do the online processing, so preprocessing images will help a lot. In _particular_ if you use Python.

Simon-Ince, 2019-03-03:
I got this working, I've posted my .py in a Gist here if it helps anyone: https://gist.github.com/Simon-Ince/910c440eb34bb722afa90853ecfb1d85

It needs some tweaking, gifs have timings for each of thier frames and I haven't taken this into consideration.