hzeller/rpi-rgb-led-matrix #627: Problem with commands not found, and or no path

TheBlueNight360, 2018-05-27:
Hi I am having a problem, where the commands that I try and not working and it says that the command is unknown. 
[image]
[image]

hzeller, 2018-05-28:
read the error message - it says that you have to go to `examples-api-use/`

TheBlueNight360, 2018-05-28:
Hi I am struggling to find how to add images and or Gif files and how to play them on the matrix. Could ypu please help and thanks for the help on the previous response

closed 2018-05-29 by TheBlueNight360

angelogoncalve, 2018-05-29:
See this file the methods:

load_animation; load_animation1

[image_utils.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/2051375/image_utils.zip)