hzeller/rpi-rgb-led-matrix #631: Project - Scoreboard

Beersatron, 2018-06-01:
First, this is a fantastic library and it made my dream of making this project a reality! Thank You @hzeller !

Second, c++ was interesting to delve into after 10+ years of developing solely in Coldfusion, a tag based website/server-side language. Strict variable types and pointers/references .... ugh!

------

**My Project:**
Interactive scoreboard for the Australian Rules Football team that I play for (Houston Lonestars). Currently viewable up to 50 yards away if you squint a little bit. Seems to run well out doors in 80+ Texas heat - under a pop-up gazebo.

**Parts:**
Raspberry Pi 3
Adafruit RGB Matrix HAT
32x32 Matrix Panels (8 of them so far)
Many 10amp power bricks (I need a better power system)
Old UPS from a server (powers everything for about 2hours)
Extruded aluminium, single and double slot
Brackets, machine screws, washers, nylon spacers and duct tape

**Code:**
I based the scoreboard code off the clock example, I will upload the file when I get to my home computer. Hopefully people can review it and give feedback on who terribly inefficient it is, I am fairly certain it is a hack job - but it works!

**Usage:**
I defaulted the wireless connection on the Pi to my phone's hotspot. I then installed VNC on the Pi and VNC Viewer on my phone. When at the field for a game I can just login to VNC and use my phone as the display. I also have a small bluetooth keyboard with track pad which is the controller for the scoreboard (each key is mapped to an action).

When I fire up the scoreboard I have it prompt me for:
Team One name
Team Two name
Team One's score (0 at the start of the game)
Team Two's score (0 at the start of the game)
Period of the game (1st, 2nd, 3rd, 4th)
Type of Period (QTR or HALF)
Number of seconds on the stopwatch (0 at the start of the game)

Within the "while (!interrupt_received)" loop I call DrawText 7 times for all this information, each one has a specific X/Y coordinate. Not sure if that is efficient? It works well and looks good. I also listen for keyboard input, and have actions mapped to QWER and ASDF that raise and lower each part of the score for each team. "P" on the keyboard will (un)pause the stopwatch.

**Next steps:**
Install a http server (Apache?) and create a web interface that would control everything, instead of the keyboard interrupts.
Make every aspect of the display editable during runtime. Right now you have to restart in order to increment the Period and there is no easy way to amend the stopwatch's elapsed time.
Add more panels so we can increase the font size.
Build an enclosed frame.

**Pics:**
Simple command line GUI
[image]

First real test
[image]

Real world usage during a club Metro game
[image]

maddogs, 2018-06-01:
I made something similar. To control I installed Apache, created a basic HTML frontend that simply updated text files that where read every second.

To control I configured the Pi itself to be the hotspot and my phone connected to that, then opened http://localhost  to access the front end and update the teams and scores.

Video here

https://youtu.be/00ndr0jZcAM

Beersatron, 2018-06-01:
@maddogs that looks awesome! Local football team/pub league?

What panels are you using? What setup did you use to power everything? I have plans to incorporate our sponsor logos into the design and installed Gimp on the Pi to test some simple resizing, yours looks very sharp - especially with the transition effects.

Are you using images or text or a combination? I'd be interested in reading your code if it is shareable.

maddogs, 2018-06-01:
Local Non League football. The panels were Indoor 32x64 ones from AliExpress.

The sponsor and advert images are exact pixel size for the panel, so no resizing required.

The score is rendered by parsing text to an image so I could generate any font size I needed.  I think i used it else where too, I can't remember.

This was all running on a original Pi3 and Jessie Lite and Python 2.7

I've uploaded the source here

https://drive.google.com/file/d/1-bCNkkpb9nA7IX107SFJGG9HcNVuQ5bl/view?usp=sharing

This includes the Python, HTML, text files etc. You'll have to excuse my HTML, CSS, PHP And Python script as this was my first real attempt and learning. Commenting is poor! 

I can upload a complete PI IMG file if it helps

Cheers!

Beersatron, 2018-06-01:
Thanks! I will look through the source for inspiration.

I have been getting my panels from Ada at $50 a pop. I might switch to AliExpress now that I see how much cheaper they are, even with the shipping cost. Did you have any issues with the panels from AliExpress - either hardware or software? I am already invested in the Ada panels, but if I can get some 32x32 P5 panels from AliExpress which I can pair with them ....

dzshanno, 2018-06-02:
That looks amazing - I'm currently trying to do something similar for my Hockey club - Can I also download for inspiration?

A copy of the Pi image would be great too.

David

Beersatron, 2018-06-02:
@maddogs the Pi image would help a lot!

Beersatron, 2018-06-03:
I have uploaded the scoreboard.cc in a zip file and added it my original post. Like I said ... it is a hack job! I was up until 2:30am on our first home game day trying to get something to showcase.

@openscoreboard that looks cool! Is it for displaying on a TV/Monitor?

openscoreboard, 2018-06-03:
Yes, run in raspberry with desktop, pc windows and pc linux.
You can used vga or hdmi (monitor or tv)

You can download source or binary here....

https://sourceforge.net/projects/open-scoreboard/files/?source=navbar

Beersatron, 2018-06-04:
@maddogs when you have the time, could you share how you built your frame to hold all the panels in place? The extruded aluminium that I am using has the narrowest of gaps between rows which isn't a problem until I do one big image.

maddogs, 2018-06-05:
@Beersatron  @dzshanno Sorry for the delay, i've been away this weekend

The image file is here

https://www.dropbox.com/s/ojkll0titruk720/20160926%20-%20Jessie%20Light%20-%20WiFi%20HotSpot%20SAMBA%20CGI-BIN%20LED.rar?dl=0

A few notes

- Its running a very old version on the matrix code.
- It will auto start on boot. At command enter **sudo supervisorctl** to stop etc
- Logon User name and password are default pi\raspberry
- I cant remember the hotspot password to gain access to the web interface, but you should be able to get the passphrase from cmd **nano /etc/hostapd/hostapd.conf**

I don't have a spare pi to load and check my self at the moment.

@Beersatron the LED Matrix i purchased came with 4 magnetic screw mounts per board. I found 6 steel strips and the stuck to that with a sheet of slightly darkened PVC over the top.

Beersatron, 2018-10-03:
What is the Patent number?

mbates14, 2018-10-03:
I don't know, I will have to ask and get it/dig it up. 

Personally, I dont care what people do. I love doing this kind of stuff as most people. But my company on the other hand just like all of corporate america, might care.

closed 2025-11-27 by marcmerlin