hzeller/rpi-rgb-led-matrix #636: High Amperage Power Supply Concerns / Question

gregoryz7, 2018-06-08:
Not an issue more of just a question to be safe.

I'm working on creating a display with multiple panels. I have a 60A 5V power supply that I have been thinking about using as overkill due to the high peak amperage these led matrix boards can draw [(4A for a 64x32 panel)](https://cdn-learn.adafruit.com/downloads/pdf/32x16-32x32-rgb-led-matrix.pdf). 
I want to make sure that the active board that you have provided in your repo (thank you so much for all of this) has a means of protecting the traces / pi board when using with high amperage supplies. 

From my simplified understanding of this, It should not be a concern normally because each component will only consume the amperage that it requires but i wanted to double check if using the GPIO to power the pi (which i have done numerous times in the past) was a proper way of supplying power to the rpi3. 

From these articles [1](https://raspberrypi.stackexchange.com/questions/1617/how-do-i-supply-power-through-the-gpio), [2](https://www.raspberrypi.org/forums/viewtopic.php?t=16295), [3](https://www.modmypi.com/blog/how-do-i-power-my-raspberry-pi) I have become more concerned and wonder if a fuse should be placed on the active board to protect the pi from overloading?

For instance: If 4 panels are on at full brightness white 4A*4 = 16A draw; Could the pi 5V line trace be at risk burning?

Concern coming from article 1:
> No - Image connecting a 5v 5A power source to the Pi and using the 5v rail somewhere to power motors for example @ 4AMP draw. the 5v trace will burn! Not the chip. The chip happily draws it ~500ma. If there was no fuse rated at 1A at the USB port and you supplied 5A via it- A short circuit will melt the 5V traces. That is why its there- to protect the PCB form overload and protect the CPU/GPU from over voltage(but we are not concerned about voltage here). PS I did not down vote you. – ppumkin Aug 16 '12 at 12:37  

Fuse question/suggestion coming from article 2

hzeller, 2018-06-08:
The Pi will only draw whatever it needs, say 500mA, so the trace on the board only needs to be able to withstand that.

The actual panels are powered by your wiring that you put directly on the panels, so you don't have to worry that the copper trace will have to carry more (I suppose you also read the [A Word About Power](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/wiring.md#a-word-about-power) paragraph in the wiring page).

There can be other problems: with high current supplies can be that they have trouble to regulate quick changes in load changes (e.g. switching from displaying black to white), so there might be brief voltage spikes that dip the voltage to 4.5V or increase to 5.5V or so. So make sure to have enough capacitors close to the LED connectors to not have to worry about that.

So nothing to worry about. You can power your Pi with a 1000A supply and you're still good (though overkill hehe).

gregoryz7, 2018-06-08:
Thanks for the quick reply, Thanks for pointing out the wiring.md page, I actually forgot you had that. 

So the persons comment in article 1 is just incorrect? Basically since the 5V rail of the Pi is not in the path of the motors load a high amp draw would not go through the traces on that board. Correct?

Internet search tangents can instill a lot of doubt :) 

And I will incorporate those caps!

hzeller, 2018-06-08:
Well the article 1 describes a scenario in which you power the Pi and then connect something else on the 5V rail of the Pi to power something heavy, so you'll have to carry all the current through the Pi to your heavy load.

But in our scenario here, we have the heavy load (LED-matrix) directly connected to the power supply (in fact, there is no way to power the LED panels through the Pi, as the IDC connectors don't have a 5V rail). The 5V is only there to power the Pi.

The only think to worry about is shorts: if you accidentally drop your Pi on something metal and the 5V rail shorts with GND, then you have the scenario of a very heavy load (metal piece) being powered through the Pi, melting traces - because your Power supply is happy to pump in more and more current and satisfy anything up to 60A.
For that particular reason (accidental shorts), having a fuse between the power-supply and the Pi is still a good idea.

gregoryz7, 2018-06-08:
Makes sense. Thank you for your time. Its time to get that 1000A supply now :)

closed 2018-06-08 by gregoryz7