hzeller/rpi-rgb-led-matrix #637: Image intermittently distorted

danielhdzzz, 2018-06-13:
I set up a 30 panel display (3 chains of 12 panels) but the image is quite distorted. I’ve been trying to fix this by tweaking --led-pwm-bits, --led-pwm-lsb-nanoseconds and --led-slowdown-gpio.

The best results i got so far are with the following settings:
--led-rows=32 --led-cols=64 --led-chain=12 --led-parallel=3 --led-slowdown-gpio=2 --led-pixel-mapper="U-mapper" --led-show-refresh --led-pwm-lsb-nanoseconds=1000 --led-pwm-bits=1

This is how it looks: [https://www.dropbox.com/s/3tibb4c2edxrb8p/IMG_3679.MOV?dl=0](https://www.dropbox.com/s/3tibb4c2edxrb8p/IMG_3679.MOV?dl=0)

Sometimes the image displays properly, but it gets distorted intermittently, especially towards the end of each chain. I can’t seem to figure out why this is. I’m running a simple python script that displays images and text (with GPIO slowdown). If I run one of the C++ examples, it flickers a tiny bit, but it’s very minor.

Any idea why this might be happening? I'm very confused.

Thanks

danielhdzzz, 2018-06-14:
Thanks for your reply. I've noticed that when I run stuff in C++, it let's me increase the --led-slowdown-gpio up to 4, but whenever I try with python, it seems to be capped at 2. Any way to get around this?

angelogoncalve, 2018-06-15:
Dear Daniel. If you have a raspberry pi 3 you can increase the slowdown for the gpio using python doing this:

from rgbmatrix import RGBMatrixOptions
options = RGBMatrixOptions()
options.gpio_slowdown = 4

danielhdzzz, 2018-06-17:
Many thanks for your help. I tried, adding these settings to my python script (and not setting --led-slowdown-gpio as a command line flag) and it didn't seem to make any difference. The result is exactly the same as setting --led-slowdown-gpio=1, which makes me think that for some reason it's not working. Why do you think this is?

angelogoncalve, 2018-06-17:
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import time

number_of_rows = 32
number_of_panels = 12
parallel = 3
number_of_columns = 64

def rgbmatrix_options():
    options = RGBMatrixOptions()
    options.multiplexing = 0
    options.row_address_type = 0
    options.brightness = 100
    options.rows = number_of_rows
    options.cols = number_of_columns
    options.chain_length = number_of_panels
    options.parallel = parallel
    options.hardware_mapping = 'adafruit-hat'
    options.inverse_colors = False
    options.led_rgb_sequence = "RGB"
    options.gpio_slowdown = 4
    options.pwm_lsb_nanoseconds = 130
    options.show_refresh_rate = 0
    options.disable_hardware_pulsing = True
    options.scan_mode = 1
    options.pwm_bits = 1
    options.daemon = 0
    options.drop_privileges = 0
    options.pixel_mapper_config = "U-mapper;Rotate:0"
    return RGBMatrix(options=options)

display = rgbmatrix_options()

danielhdzzz, 2018-06-19:
This did the trick. Many thanks!

closed 2018-06-25 by danielhdzzz