hzeller/rpi-rgb-led-matrix #638: How can I put a timer on the demo Clock?

Spazmic, 2018-06-14:
Having a hard time finding a way to replace the -t flag in python. Any idea?

GSheltonIII, 2018-12-12:
I think that you and I are looking for the same thing. Did you ever find a solution? 

I am displaying a series of gifs and want to show the clock for a few seconds before the loop starts again. But I don't want it to get stuck on the clock. I am very new to all of this so writing my own clock is not an option at this time.

maddogs, 2018-12-12:
Something to get you started would be to get the time, draw it to the canvas, sleep for a bit, then refresh and loop back round.

    matrix.Fill(0,0,0)
    currenttime = datetime.datetime.now().strftime("%H:%M:%S") 
    graphics.DrawText(matrix, Threefont, 4, 27, blue, currenttime)
    time.sleep(0.3)

I'm still using an very old version of this repo, as it works. So there may be a better way of doing this now.

GSheltonIII, 2018-12-12:
Thank you. I will play with that and see what I get. :)

Spazmic, 2018-12-13:
That's how I did it with subprocess and psutil. I created a little function to show clock and called it from my main.

```def showClock():
    proc = subprocess.Popen("sudo ./examples-api-use/clock -f fonts/7x13.bdf --led-chain=5 --led-rows=16 -d" + "%T------%d/%m/%y" + " -C 180,0,0 -O 0,0,0 --led-brightness=1", shell=True)

    clocktime = 30
    print(("Show clock for " + str(clocktime) + " seconds at: {}\n".format(time.ctime())))

    for _ in range(clocktime): # 30 seconds
        if proc.poll() is not None:  # process just ended
            break
        time.sleep(1)
    else:
        # the for loop ended without break: timeout
        proc = psutil.Process(proc.pid)
        for child in proc.children(recursive=True):  # or proc.children() for recursive=False
            child.kill()
        proc.kill()
```

GSheltonIII, 2018-12-13:
Sweet! Thank you. I will give that a go. I will let you know if I have the same issue with the stopping.