hzeller/rpi-rgb-led-matrix #642: Multiple boards in different orientations

dzshanno, 2018-06-20:
HI,

Is it possible to rotate only some panels in a chain?

I am making a scoreboard

I have 4 panels and want to have 2 joined  in landscape mode (to show the time or timer) and  2 in portrait mode to show the scores

Is this possible? Or is there a way to rotate text?

D

dzshanno, 2018-06-20:
An example of what Im trying to do

[image]

angelogoncalve, 2018-06-20:
--led-pixel-mapper=Rotate:90

dzshanno, 2018-06-20:
Thanks - But that rotates the entire matrix doesnt it? I only want to rotate the text on 2 of the 4 panels

hzeller, 2018-06-21:
You can have the LED matrix map pixels in any way you want, by implementing a [`PixelMapper`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/pixel-mapper.h). Here, you only need to wrap your head around the pixel mapping once (what visible pixels shows up on what matrix coordinate system).

Since your case is pretty special, there is of course no pre-defined pixel mapper yet, but [lib/pixel-mapper.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/pixel-mapper.cc) might give you some inspiration.

Also see the [Remapping Coordinates](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#remapping-coordinates) section in the documentation.

angelogoncalve, 2018-06-21:
from PIL import Image

    
def open_image(image_path):
        image = Image.open(image_path, 'r')
        if image is None:
            return
        return image

    
def rotate_image(image, angle):
        return image.rotate(angle)

dzshanno, 2018-06-21:
Many thanks - got it working using images instead of fonts

[image]

Beersatron, 2018-06-21:
Great! Can you drop a code snippet?

angelogoncalve, 2018-06-22:
https://github.com/dzshanno/RpiScoreboardV1

solecize, 2018-07-11:
Can you use a chain of boards with different pixel sizes? For instance I need a marquee that is 32 high and 112 wide, so I would need 3x32x32 plus a 32x16.

closed 2018-08-02 by hzeller