hzeller/rpi-rgb-led-matrix #643: Matrix only showing red channel for all pixels

ajbowler, 2018-06-22:
I've got a user reporting that their matrix is only showing the red channel (no green or blue) on their matrix. The software they're running is [my baseball scoreboard](https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard), which is using the Python bindings with this project as a submodule.

It looks like this:

[image]

A working example of what it should look like, for reference:

[image]

Curiously, his board is able to run the graphics.py sample in this project and it shows greens and blues just fine.

This may just require him to supply some additional command line flags but I admittedly don't have a lot of experience with this at the hardware level so I could use some guidance on what to look for here (a multiplexer mapping, maybe?)

His board: https://www.aliexpress.com/item/High-Refresh-Rate-Indoor-P6-RGB-SMD3528-Full-Color-32x32-Pixels-LED-Module-192x192mm-for-P6/32763763208.html?spm=a2g0s.9042311.0.0.40694c4dPUdfPH

He's using an Adafruit Bonnet (same wiring scheme as the HAT) with `--led-gpio-mapping=adafruit-hat --led-gpio-slowdown=2` as his current parameters.

The original issue on my repo: https://github.com/MLB-LED-Scoreboard/mlb-led-scoreboard/issues/180 It contains some additional screenshots of the underside of the board which may help diagnose this. I see a wiring table next to the cable input that might serve as some clue.

Would appreciate some help on this, it'll help me learn how to diagnose these issues based on the matrix wiring.

hzeller, 2018-06-22:
Could it be that `--led-pwm-bits` is set so low, that the blue values are under the threshold ?

In general, the output looks pretty noisy, so I'd suggest some tweaks e.g. with --pwm-lsb-nanoseconds.
Noisy output also can happen if the cabeling is too long. Possibly also led-slowdown.
Could also be, that the one-wire thing is enabled ?
In general, I suggest to go through the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) entirely to eliminate common issues.

(I also strongly suggest that the user is doing the [pwm hardware mod](https://github.com/hzeller/rpi-rgb-led-matrix#improving-flicker) to have more stable output; just one little solder joint).

ajbowler, 2018-06-22:
With the bonnets, there's no soldering. They're the same wiring as the HAT, but they're a straight plugin adapter to the GPIO pins, no soldering like there is with the HAT, so the pwm hardware mod isn't doable in this scenario unless we gut the adapter (and whatever warranties Adafruit gave!)

We'll play around with those flags and run through the troubleshooting section and report back, though to my knowledge, pwm-bits is running at the default (11).

hzeller, 2018-06-22:
You can still connect GPIO 4 and GPIO 18 directly on the soldering points on the GPIO header (that would be connecting pin 7 and 12).

ajbowler, 2018-06-26:
Changing around the pwm-bits appears to be having an effect. Strangely, passing in the flag with 11 as the value made it start rendering greens as well as reds, when 11 should be passed in by default. Need to dig into my Python code and see if the 11 is getting set by default or not.

Still no blues, though. I will close this issue and reopen if I get stumped.

Depending on resolution I might make a wiki page somewhere with some LED flag combinations for specific boards as this one is a special snowflake.

closed 2018-06-26 by ajbowler

hzeller, 2018-06-26:
Might sound like a mismatch between the Python code and the c library. Did you do a clean install of the Python library at the version you are using the package ?

ajbowler, 2018-06-26:
Yes, I have an install script that runs through the Python setup documented in the Python bindings folder, including all the library compilation steps.

I'll have to dig in more on my end to see what might be happening.

sachasayan, 2020-03-19:
Just adding a possibility for anyone coming across this in the archives after doing some google-fu: Check that you don't somehow have a broken connection somewhere in your power solution. 

[image]

[image]

You're likely to get just red when you're feeding signal, but not properly feeding auxiliary power. My assumption is the signal is just enough power to light up the red component of each smd/led, but not enough power to light up the green or blue components. 

This information courtesy of a janky barrel jack in my own setup that needed to be removed from the equation. 😇

RomWar1974, 2023-03-03:
As [sachasayan](https://github.com/sachasayan) said above, red pixels = low power input. Poor soldering can do this on an Adafruit Hat repair to the barrel connector.