hzeller/rpi-rgb-led-matrix #644: How do I go about scrolling negative numbers using scrolling-text-example.cc?

D9X6, 2018-06-23:
When I try to scroll "-300"
```
sudo ./scrolling-text-example -f ../fonts/9x18B.bdf -300
```

It creates this error:
```
./scrolling-text-example: invalid option -- '3'
usage: ./scrolling-text-example [options] <text>
Takes text and scrolls it with speed -s

```

D9X6, 2018-06-23:
Awesome! Thanks a lot @hzeller 👍 
I had a feeling it had something to do with the flags, just didn't know how to go about Googling the issue.
I'll be sure to give that link a read. Thanks again!

D9X6, 2018-06-23:
Hey @hzeller, how do I separate dollar signs?

I've tried "-$300" and "--$300" but it doesn't work.

ajxn, 2018-06-23:
Look up how to quote arguments and what special characters there are in
shell command line. You might want to try "man bash" to get that, or just
Google it.

The $ character is used to get value from shell variables. Quote it with \$
or with '-character, like '$300'.

hzeller, 2018-06-24:
You really should just just implement the text scrolling directly in Python, that saves you a lot of headache with invoking some external binary.

closed 2018-06-24 by hzeller