hzeller/rpi-rgb-led-matrix #648: Yet another P10 outdoor with a different configuration

mbates14, 2018-06-27:
So, I have an outdoor P10 module that is 32x16 HUB75 but it is a front service access module that we use in our LED signs that have the Nova control systems. or linsn. 

Anyways, I tried the library out and of course its a different configuration obviously because none of the led-multiplexing numbers display right. 

Since C programming and math in general is light years above my head, I was wondering if someone could help me out here. I have sent an email to the manufacturer to see if they have a datasheet on this tile which is a shot in in the dark which may help out on how to get this working?

I attached some photos of the tile. 

Here is a picture of the tile itself:
[image]
[image]
[image]

Here is the command I ran to produce this:

sudo ./demo -t 20 --led-rows=16 --led-cols=32 --led-chain=1 --led-multiplexing=0 --led-gpio-mapping=adafruit-hat --led-slowdown-gpio=4 -D 3

[image]

Then I changed multiplex from 0 to 1 and got this:
[image]

Then I tried 3:
[image]

Then for fun, Here is 8:
[image]

So as you see, its a jumbled mess no matter what multiplex number I try. 

I then tried adding and changing row address type, and scan types rotating through the multiplexing again, and its still jumbled mess, different each time of course. 

So I am 1000% lost here, so can someone help me out?

mbates14, 2018-06-27:
Using spinning cube, and grayscale, I see 3 is the closest, it breaks the panel up into 4 sets of 4 rows. 

So I assume a normal number 3 would be 1, 2, 3, 4. or 16 pixels tall. 

However, On my tile, its showing up as 2, 1, 4, 3. So the groups of 4 are shown in the wrong spots. 

Problem is I dont know the math to re-arrange it.

mbates14, 2018-06-28:
ok, so... To teach myself a little bit of how this insanity works, I decided to use an online debugger/c++ compiler to test the multiplex translator for "Spiral" which is the closest I could find. 

Well running through loops, I got this:

Y0 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48
Y1 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48
Y2 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48
Y3 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48

Y4 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63
Y5 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63
Y6 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63
Y7 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63

Y8 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48
Y9 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48
Y10 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48
Y11 = 7 6 5 4 3 2 1 0  23 22 21 20 19 18 17 16  39 38 37 36 35 34 33 32  55 54 53 52 51 50 49 48

Y12 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63
Y13 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63
Y14 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63
Y15 = 8 9 10 11 12 13 14 15  24 25 26 27 28 29 30 31  40 41 42 43 44 45 46 47  56 57 58 59 60 61 62 63 

So looking at that, and looking at whats going on, All I need to do is reverse the condition that produces this number matrix. 

So instead of 7, 6, 5, .... being on the first line, it needs to be 8, 9, 10..... 

So I studied the formula some more, which I arrived at this point:

    cout<<((2*quarter*panel_quarter)
                 + (is_top_stripe //Test if true
                    ? panel_quarter - 1 - offset //Perform this equation if true
                    : panel_quarter + offset)) << " "; //otherwise perform this

Ignore the Cout as I was using that for debugging purposes. I swapped the True/False condition around for is_top_stripe from that, to this:

    cout<<((2*quarter*panel_quarter)
                 + (is_top_stripe //Test if true
                    ? panel_quarter + offset //Perform this equation if true
                    : panel_quarter - 1 - offset)) << " "; //otherwise perform this

That reversed my output as I expected, I havent tried it yet as I am not back at the "lab" but I dont see why it shouldnt work. 

if it does, I guess I will add a "number 10" to the multiplex.

mbates14, 2018-06-28:
Must be doing something wrong, I changed the code and recompiled. didnt change anything. So my "make" skills must be rusty, or there is another code block its running somewhere. HELP

mbates14, 2018-06-28:
Nevermind, I had to do a "Make All". Just MAKE didnt work. 

My changes are functional. :-)

davemaster, 2018-10-19:
> 
> 
> Nevermind, I had to do a "Make All". Just MAKE didnt work.
> 
> My changes are functional. :-)

Greetings.... Have You a picture/video of your success? Thanks in advance

mbates14, 2018-10-31:
Here you go: 

https://drive.google.com/open?id=173Unv9Tm1T2UpT4iKOwIBZ7nyVy3Rw7Q

Camera did a horrible job picking up the images because of the refresh, but youll get the idea. 

It's a Team Name Caption panel I designed that goes into a scoreboard.

mbates14, 2018-10-31:
> > Nevermind, I had to do a "Make All". Just MAKE didnt work.
> > My changes are functional. :-)
> 
> Greetings.... Have You a picture/video of your success? Thanks in advance

Here you go:

https://drive.google.com/open?id=173Unv9Tm1T2UpT4iKOwIBZ7nyVy3Rw7Q

Camera did a horrible job picking up the images because of the refresh, but youll get the idea.

It's a Team Name Caption panel I designed that goes into a scoreboard.