hzeller/rpi-rgb-led-matrix #649: Lower half of 4 panels doesn't display red channel

swalsch, 2018-06-27:
[image]
The lower half of all 4 of these 32x32 panels won't light up red leds. This worked properly before, when I was just following the outdated adafruit tutorial. I then did the pwm mod on the adafruit hat, setup a raspbian stretch lite with the pixelpusher implementation of this matrix and ran the demo to check, only to find this problem. I suspect wiring, since it's unlikely all 4 panels fail in the same channel in the lower half, right? Where would I start investigating? What could be the problem?

Some stats:
pi 3
adafruit hat with pwm wiring
adafruit sourced 32x32 matrix panels
raspbian stretch lite
cloned from pixelpusher git

Thanks!
Stephan

hzeller, 2018-07-05:
Did the same set-up work without the pixelpusher and just running the examples-api-use/demo ?
It does look like some wiring problem.

hzeller, 2018-07-17:
I suppose you fixed the wiring issue ?

closed 2018-07-17 by hzeller

swalsch, 2018-07-18:
So after checking and replacing some wiring and the problem persisting (yes, also in demo mode for the examples) I ordered new panels, which took a while. Sadly they show the same behaviour, which has led me to believe the adafruit hat might be faulty. So next step I'll solder the signal cable straight to the pi to rule that out.

swalsch, 2018-08-04:
I did the above mentioned step of soldering the signal cable directly to the pi, skipping the adafruit HAT and now it works!! So the R2 signal path on the HAT is somehow broken. I only use the HAT for power distribution now and everything works fine.