hzeller/rpi-rgb-led-matrix #650: U-Mapper that doesnt require flipping the tile upside-down?

mbates14, 2018-06-28:
Is there a U-Mapper that doesn't require the tile be flipped upside down? 

I just got my P10 outdoor tile working, but its a louvered tri-LED tile that you cant really flip around without it looking like poo. 

So, I am trying to build a 6 long string which is 3 tiles on top, 3 tiles on bottom. So basically, a U-Mapper. 

But it flips the bottom. 

Thoughts?

hzeller, 2018-06-29:
I'd probably just run a server (such as FlaschenTaschen) that receives data via a network protocol and send data from your language of choice. That is typically much easier than to try link these libraries on the C-level.

mbates14, 2018-06-29:
That wont work in the environment that I am working with. 

Tablet -> Bluetooth -> R-Pi. 

Using a tablet running an Android program to talk to the LED array. Sure I could use Python or anything else, but my Dev environment is B4X which has an Android compiler as well as a regular cross-platform java compiler. And cool thing about that is I can establish communication between the two running programs and "physically" send full objects between the two. So I could wrap up/serialize a Canvas object, transmit it, receive it on the other end and write it to the display. Just for example.

MarcKarasek, 2018-06-29:
I have a we server plus client on Android to do the demos.   It maybe
useful as a starting point for your development.  I am on the process
(slowly) of pulling this back into hzeller project.  It is currently in my
github.  The srvr uses udp for control plans and sends the canvas over
TCP.

If you want I can send you a link.  Or just search for Marc Karasek... My
github name.

If you have any questions just holler.

mbates14, 2018-06-29:
I like that idea and thought. I may revisit that project in the future, however I never touched native android or java. 

I am a VB developer and thats the environment that I came from, which is why I stuck to B4A like glue. its more of a transpiler than a compiler. Might want to look into it. However the native java nerds can develop libraries for it along with wrappers so there is a plethora of those. Just not for this project yet.

closed 2018-07-05 by mbates14

hzeller, 2021-10-13:
The V-mapper is probably what you guys want
https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/examples-api-use#v-mapper-and-vmapperz-vertical-arrangement

hzeller, 2021-10-13:
what are the command line options you give ? Looks like you need something like `--led-cols=64 --led-chain=2` as you wire two 32x32 in sequence.

Also, turning the final panel you then can do by using the rotate transformer.

kilowody, 2021-10-13:
sudo ./demo --led-rows=32 --led-cols=32 --led-slowdown-gpio=4 --led-gpio-mapping=adafruit-hat-pwm --led-chain=4 --led-pixel-mapper=U-mapper -D0

^this is the command line 
i am using 4 32x32 panels all connected with hub75 most similar to the U-mapper example but i am not rotating the top two panels physically

kilowody, 2021-10-13:
just tried it, that seemed to do the trick! 

thank you so much for this library btw its been super helpful!