hzeller/rpi-rgb-led-matrix #651: Brightness jumps up and down

mbates14, 2018-07-04:
I noticed when I leave the pi sit on a demo, say the Simple Square demo., the brightness pulses up and down once a second. 

Using the adafruit hat, and I did the mod to it and changed it to adafruit-hat-pwm and it really didnt make a difference. 

This is driving me nutso. any ideas? I tried isolcpu command as well.

hzeller, 2018-07-05:
Did you read [the troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting) that goes through a couple of topics what to do here ?
You probably have some process running that uses memory bandwidth, 'stealling' it from the rgb-matrix.
If it is once a second, maybe you have a `top` running somewhere ?

You can also even out time variations by measuring and setting the `FIXED_FRAME_MICROSECONDS` value in the Makefile and re-compile.
  https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105

mbates14, 2018-07-05:
Yes I did, that was the first thing I did. I am running the Raspbian lite Jessie with the sound turned off and some other things turned off. 

I dont know what top is so I am unsure how to proceed there. Maybe I need to drop the isolcpus down to 2? but I dont see where the core gets assigned to the rgbled matrix as a dedicated core unless its compiled in somewhere. 

Also I can stop it by changing the microseconds flag to around 550 or so at the command prompt, but when text is present it brings in alot of sparkling/pixel mal-discharge which I dont know whats causing that. I can get rid of most of this by going up to 1000 or so but it seriously affects the framerate. 

I am only using a single chain of 8 16x32 modules and thats it. I have it formatted as a 128x32 display using the S-mapper mod that the other guy posted.

hzeller, 2018-07-05:
The isolcpus value needs to stay at 3 as it is the core we want to isolate and run the update on.

There is no microseconds flag (there is only the PWM nanoseconds, but that is something different). If you want to eliminate the regular flicker, you need to [follow the instructions in the Makefile](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105) and modify the compile-in value of `FIXED_FRAME_MICROSECONDS` that suits your set-up.

mbates14, 2018-07-05:
ok, thank you for the information. Greatly appreciated.

mbates14, 2018-07-05:
So I did what you mentioned, my longest time was 7790. So I chose 8000. 

I did a sudo make clean all, Uncommented that line and changed it from 5000 to 8000. 

Did a save, then sudo make all. It made no difference, its still trying to show 6XXX usec and then jumps to 7790 again, and I see the brightness flickers again. So its almost as if that flag did nothing? 

thoughts?

hzeller, 2018-07-05:
So if you have changed the setting, you would see it show up in the compilation lines, e.g here you see the 8000.
```
  g++ -I../include -Wall -O3 -g -fPIC -DFIXED_FRAME_MICROSECONDS=8000  -DDEFAULT_HARDWARE='"regular"' -Wextra -Wno-unused-parameter -fno-exceptions -c -o led-matrix.o led-matrix.cc
```

I suggest not to run `sudo make all`, but only `make` or `make all`. With the 'sudo', you are writing files that can not be removed by the regular user. Use 'sudo' only for make install
(to fix the outputs of `sudo make`, you should run `sudo make clean` in the toplevel, so that all the files written as root are removed. Then just type `make` again).

Anyway, once you have compiled that, the output of `--led-show-refresh` should never really show numbers below the chosen 8000, if so, something went wrong in the compilation process and you should make clean first.

mbates14, 2018-07-05:
Thanks for the info. Is there any way to make this a runtime flag? I keep having to circle back around to it over and over. 

I am at almost 9000 and its stable until I start a text/picture loop. When it goes to load the PNG file it flickers, and jumps past 9000 when it does.

closed 2018-07-06 by mbates14