hzeller/rpi-rgb-led-matrix #652: Trouble displaying special characters (UTF-8)

D9X6, 2018-07-05:
I use a script to generate a .ppm image from a string. It then scrolls the image across the LED display using one of the supplied demos.

I specifically use the  `draw.text(pos, colour, textString, font)` python function.

I'm having trouble displaying special characters.

I have luck displaying "µ" using `'\xB5'`
But I have trouble displaying "₿" when I use any one of the following `\x83 \xE2\x82 \xBF` - either a question mark with a box around it appears, an upside down question mark appears, or an accented letter appears.

Reference: http://www.fileformat.info/info/unicode/char/20bf/index.htm

[image]

hzeller, 2018-07-05:
You might need to add the character to the font you're using. If you grep for ENCODING in the supplied fonts, you see that non of them encodes character 8383, so you'd have to manually add it.

Given that this is essentially just a B with little lines above and below, it should be fairy straight-forward to manually copy a B and modify it accordingly. You can use a font editor, or just manually do it, the format is a super-simple ASCII hex format, so you can do that by hand as well.

If you are modifying some of the included fonts, please send a pull request so that others also can make use of it.

D9X6, 2018-07-07:
Thanks @hzeller!

I ended up downloading fontforge and editing a .ttf font that I am using, to support that character now.
I never knew how technical the world of fonts can get.

hzeller, 2018-07-07:
Yes, there is a lot in fonts, in particular in vector fonts there is a lot of internal technicalities going on including virtual machines.

Anyway, glad you got it working!

closed 2018-07-07 by hzeller