hzeller/rpi-rgb-led-matrix #653: Help About Library

AKKaygin, 2018-07-05:
Like I said, I need help understanding this library. So far I ran the examples without any problem. Now I need to make a application like an andon panel but my knowledge about python and c is very limited. So I'm asking you to help me understand the basic commands (not the --led-chain stuff) of a program. Just enough to make the panel write static text according to  given variables.
Any help would be appreciated. Thanks.

--And I know that is not a kind of issue I should post here but I couldn't find any place better than here :(--

mbates14, 2018-07-05:
I found it easier to use Python for text, or images. etc. Worked fine for me. 

But the setup is super sensitive to CPU pre-emptive usage so your programs have to be tight. or itll flicker like a bat outta hell and not be very useable.

AKKaygin, 2018-07-06:
@hzeller firstly I thank you for your extreme quick response,
I greatly agree, programming, at least editing, requires some programming language knowledge. But I mastered at web development and thats why they want me to do this job, get the info from sql and write it. Unfortunately I get nothing from this:
  rgb_matrix::DrawText(canvas, font, x, y + font.baseline(),
                         color, outline_font ? NULL : &bg_color, line,
                         letter_spacing); 
So I'm asking you(pulural) to comment theese out.

hzeller, 2018-07-06:
The line will write the text to the canvas. Not sure what you mean with commenting it out, I don't have access to your source code :)

AKKaygin, 2018-07-06:
@hzeller just 
offscreen_canvas.Clear()
            len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text) //1. funtion doses this, 2. this, and 3. this ...

hzeller, 2018-07-06:
Did you then switch the offscreen-canvas to display with `SwapOnVSync()` ? Otherwise it will of course not be displayed yet.

AKKaygin, 2018-07-06:
@hzeller I'm embarrassed because I'm making you write to me _ouf_
I'm trying to do this kind of thing. I just need to write some static text with changing color. 
[image]

Actually more I think about code, more I get it. Now I just need to learn how to adjust position and write multiple lines. 
 len = graphics.DrawText(offscreen_canvas, font, pos, 10, textColor, my_text) which function adjusts the position of the text?

hzeller, 2018-07-06:
The pos and 10 is the x and y coordinate in your case.

The [header file](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/include/graphics.h#L83) has all the description for this function.