hzeller/rpi-rgb-led-matrix #654: Color Mapping Error

loewek3628, 2018-07-06:
Hi,

We currently have a display with 3 parallel chains of 6 matrices each.

Here is what the display shows:
[image]

Here is the original image: 
[image]

We think our issue is with our second chain of wiring. There are clearly color issues and some sort of duplication of the image.

As of now we have tested and have found no issues:
- Continuity of the wires
- Power
- Different ribbon wires
- Different matrix panels
- Multiple different images and text strings

Any other ideas to troubleshoot our problem?

hzeller, 2018-07-06:
It looks like that in your middle chain the green and blue of the first and second half-panel are swapped. G1 and B2 or something along those lines. Double check that you have all R1,G1,B1 and R2,G2,B2 at the correct places. Did you double-check that there are no solder-bridges on the adapter board you soldered ?

So if you have a test-image that shows just red, green and blue patterns it might be easy to spot.

loewek3628, 2018-07-06:
We tried showing a RGB only test image and there was no visible error which seemed odd.

Switching the pins around was also unfruitful because even though we could see the change in colors, it didn't change the strange splitting or doubling effect. 

Currently, the wiring is done with male-to-female jumper wires so the only soldered connections are for the A, B, C, D, Strobe, etc. pins and everything there shows connectivity via the multimeter.

Here is what scrolling text that is supposed to be white (and not doubled) looks like:

[image]

loewek3628, 2018-07-06:
Whoops! You were right - Green 1 and Green 2 were switched....

closed 2018-07-06 by loewek3628

hzeller, 2018-07-06:
in the above image, is the color for the top-text white or is it magenta ?
It sounds like for the middle chain, the greens G1 and G2 are switched.