hzeller/rpi-rgb-led-matrix #655: Set brightness without rewriting the canvas?

mbates14, 2018-07-06:
I am despirately seeking a way to dynamically set the display brightness while the existing contents are on the display. 

Of course matrix.brightness doesnt do anything until I redraw the canvas. Thats not good. 

So as a workaround, I tried creating a global variable of DrawCanvas as CreateFrameCanvas(). 

I can draw into that, and then do SwapOnVSync in a function and it works fine. 

But if I simply change the brightness values, and re-run SwapOnVSync using the same variable DrawCanvas, nothing happens. 

any ideas?

hzeller, 2018-07-06:
you essentially have to keep the content separately in an array somewhere and redraw a frame with that content. You can simplify that by implementing a canvas interface that delegates to an underlying canvas, but also keeps the rgb values, then a function that dumps these to the underlying canvas if needed.

mbates14, 2018-07-06:
Yuck. Sounds messy. 

What happened to just simply setting the brightness and magic happens?

mbates14, 2018-07-06:
> You can simplify that by implementing a canvas interface that delegates to an underlying canvas, 

How do I do that?

hzeller, 2018-07-06:
That magic essentially would require what I just described :) Since not everyone needs that, it would require to use more memory in case it is not needed.

Background: the underlying canvas already has to preprocess the data in a way that it can be written out quickly to the display, so the memory layout already is in a way that doesn't allow easily to change the brightness later.

mbates14, 2018-07-06:
not everyone =/= no one. That statement drives me nuts. No offense. 

The way I see it from what you described, setting a brightness value should execute a function that re-renders the "memory layout" with the new values IMHO.

hzeller, 2018-07-06:
Yep.

mbates14, 2018-07-06:
If memory is a concern then live brightness changing could simply be excluded with a makefile flag just like the other features are currently. IMHO. 

But I cant fix it as I am not a C programmer. So please. can someone?

hzeller, 2018-07-06:
It is not only memory, it is simplicity of the library. Adding kitchen-sink features makes it hard to use and understand.
Anyway, if you feel strongly that other people might need this, you can provide it as an additional header file with sufficient documentation and send a pull request. We're all in this together, features only happen if someone needing it implements it :)

mbates14, 2018-07-06:
Sadly I am not a C programmer so that'll never happen. I was hoping you as the author would probably do it. But I guess I was wrong. 

For now I will work around it I guess. Screw it.

hzeller, 2018-07-06:
I could probably do it, but you'll have to wait for the weekend as I am at work currently.

Make sure to watch your tone if you want others to spend their free time to work on your project...

mbates14, 2018-07-06:
Excuse my tone. Just frustrated for multiple reasons which is not yours, or anyone's fault. Sorry if I was being offensive.

For now I just added a brightness value check before I render each frame. That works for now.

closed 2018-07-17 by hzeller