hzeller/rpi-rgb-led-matrix #661: Question about mirroring panels.

mattyams, 2018-07-14:
I am trying to put the same image on 6 panels and I am using the Active-3 adapter board. I have 4 32x32 boards wired to Panel 1 and 2 32x32 boards wired to Panel 2 on the Active-3. I just can't figure out how to get a 32x32 gif on all 6 boards.

mattyams, 2018-07-27:
Just to add, I am making a battery powered cube of 6 32x32 panels. I have it working if I make an image containing 6 images.  But I want to be able to run gif files on each panel.

hzeller, 2018-07-27:
Running gifs is the same, you essentially just set the pixel on each panel on the same position when you fill the canvas.

There is a simpler hardware way if you always want all panels to be mirrored: just connect them in parallel to a single output. So instead of a chain, you connect all to the same output, then just send an image with `--led-chain=1 --led-parallel=1`. All panels will show the same.

mattyams, 2018-07-27:
Perfect! Thanks for the help.

mattyams, 2018-07-27:
So that did work as expect but the very last panel in the chains colors seem a bit off.  I have tried a different panel, different cable, separate power and the panel by itself works when no others are connected. Any ideas?

hzeller, 2018-07-27:
sounds like you still have them all in one chain. If you do, a similar effect happens, but the colors are off as you noticed.

What I mean is: you need to connect _all_ inputs of the panel to the output on the Pi. So that all of them are connected to the same spot. Then, all of them get the same signal and display exactly the same.

hzeller, 2018-07-27:
you essentially need to make some wiring with 6 connectors, with each pin connected to the corresponding pin on the others. you need to solder something together creatively.

Doing it in software (just setting six pixels in each panel) might be simpler, depending if you are a Hardware or Software person.

mattyams, 2018-07-27:
Luckily I am both but I want to try the software solution.  How would I go about that. Right now I have the following running in my rc.local to start the cube when I press a button. 

`home/pi/rpi-rgb-led-matrix/utils/led-image-viewer --led-pixel-mapper="U-mapper;Rotate:180" --led-parallel=1 --led-chain=1 --led-brightness=25 -t5 -f -s /home/pi/images/*.gif`

hzeller, 2018-07-28:
For the software solution you 
  * either set the pixels in 6 different locations manually
  * or write a Canvas, that behaves like a 32x32, but then delegates to an underlying canvas that is 192x32 an does the setting-pixels at every x in increments of 32.

then you plug your code in the right place in the led-image-viewer.

closed 2018-08-02 by hzeller