hzeller/rpi-rgb-led-matrix #662: Where do you source your panels from? Adafruit, AliExpress, other?

Beersatron, 2018-07-15:
I have 8 32x32 (5mm pitch) panels from Adafruit but they are $45 a pop and I plan on at least doubling the size of my scoreboard project. I bought them in two batches of 4, months apart, and the second batch has a noticeable color tone difference (which isn't a deal breaker).

For the sake of consistency and cost I have been looking around for a 'better' source of panels and so far have:

- [Adafruit 32x32 @ $45](https://www.adafruit.com/product/2026?gclid=Cj0KCQjwx8fOBRD7ARIsAPVq-Nn1XUl7bwr_cMy6MY0rltnkVPe2sYNR2Ee_wXaQqGhpUOzQze15zmcaAoc2EALw_wcB)
- [AliExpress 32x32 @ $12](https://www.aliexpress.com/item/HD-outdoor-led-display-p5-full-color-module-smd-wholesale-price-p5mm-outdoor-led-screen-module/32658170357.html?spm=2114.search0104.3.1.583a42ff4NIytD&ws_ab_test=searchweb0_0,searchweb201602_1_10152_10151_10065_10344_10130_10068_10324_10547_10342_10325_10546_10343_10340_10548_10341_10545_10696_10084_10083_10618_10307_10846_10059_100031_10103_10624_10623_10622_10621_10620,searchweb201603_12,ppcSwitch_3&algo_expid=484a1331-a5d8-45da-95ba-a047ebca0b96-0&algo_pvid=484a1331-a5d8-45da-95ba-a047ebca0b96&transAbTest=ae803_1&priceBeautifyAB=0)

I am curious where others are getting their panels from and are you using 32x32 or 64x32? Have people had any issues buying from AliExpress?

Beersatron, 2018-07-16:
Would this panel work? I don't want to order 9 of them just to find that I missed something obvious ...

- [AliExpress 32x64 @ $13.06](https://www.aliexpress.com/item/Indoor-P5-SMD2121-Full-Color-LED-Module-1-32-Scan-Current-Driver-330-160MM-RGB-SMD/32636193207.html?spm=2114.10010108.1000014.4.52d1460f0xBsdL&scm=1007.13338.98644.000000000000000&scm_id=1007.13338.98644.000000000000000&scm-url=1007.13338.98644.000000000000000&pvid=2334013b-420e-4d3b-970f-c49373fa199f&_t=pvid:2334013b-420e-4d3b-970f-c49373fa199f,scm-url:1007.13338.98644.000000000000000) (SMD2121 1/16 Scan)

hzeller, 2018-07-17:
AliExpress is usually the cheaper variant with panels exactly like Adafruits (Adafruit just imports them from China and marks them up somewhat).

Almost all of the panels you find marked with HUB75 should work. The 32x64 above looks like it would.

Beersatron, 2018-07-18:
Thanks @hzeller , I think I might go for this slightly more expensive one as I noticed the SMD2121 URL has 1/32 scan in it - even though the description of the item says 1/16 I don't want to take the chance on a possible typo. Also, the Adafruit tech sheet states they use SMD3528 which this new linked panel also uses.

- [AliExpress 32x64 @ $18.01](https://www.aliexpress.com/store/product/Free-Shipping-P5-Indoor-SMD-Full-Color-LED-Module-1-16-Scan-Current-Driver-330-160MM/709678_32273502525.html?spm=2114.10010108.1000023.10.30c679347NU7Bd) (SMD3528 1/16 Scan)

Have you noticed any difference between SMD2121 and SMD3528, plus 1/16 and 1/32 Scan?

Is there anything to stop me chaining my 8 existing 32x32 Adafruit panels with the 32x64 panels I will be getting?

Beersatron, 2018-07-30:
I received my 9 panels from AliExpress and had enough 5v 10a power supplies to power 6 of them in a U in order to test. They look good!

@hzeller if you have the time, could you expand on your [A Word About Power](https://github.com/hzeller/rpi-rgb-led-matrix/blob/04625575b935de429efd9b4d4cc0143f2b87b95c/wiring.md#a-word-about-power) article please? Specifically, links to parts for the frame, how you assembled the frame (and made it conduct the power) and then also the actual power supply you used?

I currently have 3 5v 10a power bricks and will need another brick per 2 panels as I increase the overall size of the scorecard. But, I would rather use something similar to what you have in your article.

I did build [this AdaFruit frame](https://learn.adafruit.com/adafruit-diy-led-video-wall/frame) but it isn't as sturdy as it needs to be due to the small (4mm) machine screw size that my panels use, plus the spacers and washers don't quite match.

Thanks!

hzeller, 2018-08-02:
The aluminum parts are simply profile aluminum from the hardware store. They were a bit tedious to drill all the screw holes, but other than that work fine. There is no further documentation from what you see in the picture :).

If you have multiple power supplies, make sure to connect all the GNDs together, but _not_ the 5V. Each power supply supplies a subset of the panels.

Beersatron, 2018-08-05:
All 9 panels are now hooked up and powered, I just went with more 5v 10a switching power supplies for now. There is flickering but I think that is likely due to me running Jessie, with Apache/PHP (website to control the scoreboard) and a monitor hooked up by HDMI. I played around with the optional params and managed to reduced some of the flickering. I also disabled sound and bluetooth in config.txt.

closed 2018-08-05 by Beersatron

hzeller, 2018-08-05:
Make sure to connect the panels in three parallel chains if not already (e.g. using one of the adapters), as refreshing three is of course faster than 9 in one chain.

Don't run any Rasbian with a graphical UI, it will mess up your timings and result in flicker. The Raspbian Lite is usually a good choice (read the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#troubleshooting)). Just an HDMI screen on a text console should be fine.

If you generate flicker due to other loads on the machine, you could tweak things with the experimental [`FIXED_FRAME_MICROSECONDS`](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/Makefile#L105) setting in the Makefile.

Beersatron, 2018-08-06:
I have 2 parallel chains, one with 6 of the 32x64 panels and then another with 3 of the 32x64 panels and then 6 of my original 32x32 panels. I was actually surprised that this worked when I initialized with rows=32/cols=64. That setup wouldn't contribute to flickering somehow, would it?

[image]