hzeller/rpi-rgb-led-matrix #664: [Question] Including WiringPI to control unused IO pins

fxmech, 2018-07-19:
Thanks for this amazing library!
I need to make 4 64x64 1/32 panels in a U config display 16 images when buttons are pushed.
I did this by using 4 GPIO pins (unused by the panels) and using their state to create a 4bit number (1-15).
The plan is to assign each number to a specific image.
Is it possible to work with the Wiring PI library from within say the demo-main example?
I have not been able to figure out how to recompile the code.

If you can help, can you tell me if this approach can work and what is the process to say copy the example to a new file name,  include the WiringPI library within it,  and recompile it?

If any users would we willing to work with me for pay, I would like to talk... I'm an arduino programmer and this seemingly simple task is still far above my skill set.

hzeller, 2018-07-25:
WiringPi might work, but make sure to initialize it before the LED matrix.
I usually read and write the GPIOs directly.
If you're in San Francisco, maybe we can meet at Noisebridge some evening and hack on it (write me a mail).

hzeller, 2018-08-02:
Alright, added the capability to read input bits. See [examples-api-use/input-example.cc](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/examples-api-use/input-example.cc).

closed 2018-08-02 by hzeller

nyinyinyanlin, 2019-02-13:
Is there any Python binding for this capability?

hzeller, 2019-02-13:
If you write one and send a pull request, there will be :)

d3d9, 2019-02-13:
I added it here, with some other changes: https://github.com/d3d9/rpi-rgb-led-matrix/commit/8f2ee05c5aac42dd16f0a87eb5c021155acbe67f, not sure whether it's done well but I think it works