hzeller/rpi-rgb-led-matrix #666: Extra GPIO Pins with the Adapter Board

puffycholo, 2018-07-24:
Good day. A noob question. Will i be able to use other GPIO Pins if I use the adapter board provided?

puffycholo, 2018-07-24:
im only using 2 chains. where will i connect the gpio ports unused? i will be using this drive board 

[https://www.electrodragon.com/product/rgb-matrix-panel-drive-board-raspberry-pi/](url)

hzeller, 2018-07-24:
Yes, if you only use two chains, you can use the remaining GPIOs.

You find the available GPOSs by  checking out the p2_XX values in https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/lib/hardware-mapping.c

puffycholo, 2018-07-24:
I will be using 64x32 panels. Do i need to tweak the adapters just like the 64x64 board config?

hzeller, 2018-07-24:
No, 64x32 is essentially just a long 32. So no tweaking needed.

closed 2018-07-25 by hzeller