hzeller/rpi-rgb-led-matrix #667: Adafruit 16x32 RGB not working

ace16vitamine, 2018-07-24:
Hi,

I have a simple question...

I started to use the Adafruit 16x32 RGB matrix panel with rpi-rgb-led-matrix, but I can not find a how to that is working. I wired the cable from this URL: https://learn.adafruit.com/16x32-rgb-display-with-raspberry-pi-part-2/wiring-the-displays to my pi3. 

I know that this is working, I am able to show thinks from a simple python script to the panel.

But every time I try ./demo -t 10 -D 1 runtext.ppm --led-no-hardware-pulse --led-gpio-mapping=adafruit-hat

it shows my nothing, black screen.

Any Ideas what is to do?

Thanks
Stefan

ace16vitamine, 2018-07-24:
But now I see that the LEDs on the upper side are also glowing (see picture). Is that normal?

[image]

hzeller, 2018-07-24:
Make sure to use short cables and level converters for clean digital signal transitions.

closed 2018-08-02 by hzeller