hzeller/rpi-rgb-led-matrix #668: Freezing image

currentlysober, 2018-07-24:
Hi everyone,

I was wondering whether it is possible to freeze the LEDs. My plan is to measure power consumption of one single 'pixel' (one RGB dot). As the led matrix is run by multiplexing (one dot is turned on and off many times in a second) my plan is to freeze it for e.g. 1 second. 
Is there some way in the library (e.g. by using sleep() ) to freeze the image?
I will be highly grateful for the answer.

Thanks!

closed 2018-07-25 by hzeller

currentlysober, 2018-08-01:
Hi! Thank you for information. However, if I want to check if one 'pixel' is broken, I need to turn it on and measure the peak only in this very moment (or maybe I am wrong?). Anyway, is there a way to stop refreshing or not? Even if I burn a LED, I would like to try the freezing solution. Thanks!

hzeller, 2018-08-01:
unless you manually stop the loop inside the library, you can't freeze it with the default api provided.

Wouldn't you see a broken LED anyway, as it would be a dark emitting diode?

mzdunek92, 2018-08-06:
Hey, I was trying to achieve the same result few months ago. After putting `while(true){}` loop in main `UpdateThread`'s loop (after `DumpToMatrix` function) I got black screen in less than 1/30 second. So I am afraid, that it is not possible to freeze image on these panels.
I was using P6 32x32 LED panels, 1/16 scanning.

hzeller, 2018-08-06:
The LED panels often have protection circuitry that auto-disables the LEDs if there is no update.

mzdunek92, 2018-08-09:
@hzeller Have you seen a datasheet or any other source that pictures this mechanism that you describe?