hzeller/rpi-rgb-led-matrix #671: Coding difference between 32x32 and 64x64

tomasaidukas, 2018-07-28:
I wrote a piece of code which directly controls the pins and enables me to light up a particular LED. To select a row (from 0 to 15) that needs to be used on the 32x32 LED array I send a 4-bit number (for one particular half of the LED array). This 4-bit number is sent through the A B C D pins found on the HUB75 connector. 

I want to extend this to the 64x64 array in order to select a row from 0 to 31, but sending a 5-bit number and using the E line either as a least significant bit or the most significant bit doesn't work (I can only use the 64th or 32nd rows but others don't work)

Is the addition of the E line requires additional coding to be made? I just want to keep the LED on till I give an order to change it (for microscopy purposes), but I am not sure what changes when the E line is added. Any info would be much appreciated.

hzeller, 2018-07-28:
It depends on the panel. If your panel has an E-line, then it is typically on either Pin 4 or Pin 8.
There are other panels that are internally just a 128x32, so you work with A, B, C, D but the pixels are wrapped around (there are different ways; see the `--led-multiplexing` flag).
Yet there are other 64x64 panels that only have A, B (`--led-row-addr-type=1`), but sounds like you don't have this kind of panel.

tomasaidukas, 2018-07-28:
I am using the adafruit 64x64 panel with the added E line. Since I want to use it for purposes where the LED stays lit up constantly, I wanted to avoid using the libraries you have written (I use the LEDs for microscope illumination where the exposure times are 10s of miliseconds and refreshing the LEDs - as fast as they are - are increasing the exposure times). But i wanted to know what is the difference coding-wise between 32x32 (ABCD pins) and 64x64 (ABCDE) adafruit arrays.

On the 32x32 to flash a single LED array:
1. Send a number from 0-15 (to select a row) as 4bit binary number
2. Depending on the color and which half of the LED array is use send out 32 signals using R1,R2,G1,G2,B1,B1, where each signal train corresponds to one particular column.

I tried to do the same on the 64x64 array but instead of sending a 4bit number (using ABCD) I send out a 5bit number (using ABCDE) and loop through 64 columns rather than 32. I would expect to have the same results as for the 32x32 array, but I get rubbish instead i.e. I can't select the LEDs properly. Is the addition of an E line complicating things more than I expect? I don't want to use PWM or anything fancy, just flash a single LED and keep it on. Hopefully that made sense.

hzeller, 2018-07-28:
I suppse you tested that the E-line works with the rpi-rgb-led-matrix; if so it should also work if you send binary values to it in your implementation.

Howver, some panels prevent from driven with too low frequency to prevent burn-out of the LED, so that could also be the case if nothing happens. Others limit the time you can pull down the `~OE` line, so you might get shorter exposure than you want.

Why do you use a panel for the microscopy application, wouldn't a single LED be enough ? Or do you try to do some dark-field exposure from different angles ?

tomasaidukas, 2018-07-28:
Maybe it is a speed issue then. Do you have any idea what's the slowest time that one could pull on the OE line? If you believe that what I'm trying to do should work, I will look into optimizing it, though that might be challenging.

I obtain multiple images of the sample, each obtained using a different illumination angle (corresponding to different LEDs). I can then synthesize these images computationally to increase the resolution of the final image. (Here is some stuff that I have done https://www.researchgate.net/publication/324071935_Miniature_Fourier_Ptychographic_Microscope_Using_Mobile_Phone_Camera_Sensors). The 64x64 panel has a smaller pixel pitch enabling me to put the LED array closer to the sample while maintaining the same imaging parameters that I had using 32x32 array with a 5mm pitch (better signal-to-noise ratio while providing the same image resolution post-reconstruction). Unfortunately, the task of swapping these arrays is proving to be less trivial than I expected it to be.

hzeller, 2018-07-28:
It is hard to tell if it is a speed issue, it really depends on the type of LED panel; some are finicky, others are less.

Your research sounds interesting and a nice idea to computationally increase resolution.
What is the typical exposure time you need for one image ?

If you use the regular mode with 11-bit PWM, the LEDs strobe of course (which might be a problem if it is cells that react to that), but for the overall exposure time, it should be possible to just leave them on as long as you need. With a 1-bit PWM, the LED only goes on a defined time per image-refresh.
Using `SwapOnVSync()`, you can  leave an 'image' (a single lit pixel in your case) on for a define number of frames, so that might help to get the timing right.

With `--led-pwm-bits=1`, you can make sure that there is a little as possible PWM going on. In fact, with this option, and a single frame displayed with `SwapOnVSync()` you switch on an LED for a very defined time once.

The fine-tuning of how long that is could be done with `--led-pwm-lsb-nanoseconds`.

tomasaidukas, 2018-07-28:
The exposures for brightfield images are around 10-50 milliseconds, but for darkfield they can go up to hundreds or thousands. I noticed last night that with the 64x64 panel and using your code to flash a single LED I had to increase the exposure time up to 400 milliseconds (previously used 40). I still need to measure if the power of a single LED on 32x32 and 64x64 LED array differs.  If they are the same then I would have expect the exposure time to drop dramatically instead of increasing so much! So far I assumed that the power is the same and attributed the lowered exposure time to the refresh rate of the LED array.

Thanks for the suggestions, I will try it out tomorrow and do some testing! My samples are not sensitive to strobing, I just need to get the maximum power from a single LED, with as little flickering as possible. 
Do you initialize the canvas first, apply canvas = matrix.SwapOnVsync(canvas); and the setPixel? (I will try to look through your codes to see in more detail how that will work). If what you are saying would work then that would be great. Thanks again for all the help.

hzeller, 2018-07-29:
Yes, the longer exposure time is due to the multiplexing, as the LED is only on for 1/16 of the time in case of the 32x32 panel or 1/32 in case of the 64x64 panel.
So using the 64x64 panel will increase the exposure even more as the multiplexing has to cover twice the number of pixels. _Also_, the 64x64 panels typically have darker diodes (as they are much smaller).

Now, the problem is for your case: You can't leave a single LED in these panels on for hundreds of seconds, it will burn out. These LED panels are designed to send brief overcurrent to the LEDs, but since they are multiplexed, the average current is only 1/16 and they are ok with it.
But if you left a non-multiplexed LED on for more than a couple of seconds with the current they are operated at, it would soon become a DED (Dark Emitting Diode :D ).

To get more brightness, I suggest to look for so-called 'outdoor panels'. They typically come in 16x32 or 32x32 (but I have not seen them in 64x64, and unfortunately also not with smaller pixel pitch which is what you might want). They use one less address line, so the multiplexing rate is smallre (16x32 is 1:4 multiplexing, 32x32 is 1:8) (internally, they are organized as 8x64 and 16x64 wtith LEDs going in slalom over the panel; see the `--led-multiplexing` flag). Due to the less multiplexing and special diodes, the outdoor-panels are much brighter. But as well for them: you can't leave a single LED on for too long.

As for `SwapOnVSync()`: have a look at the code examples. You essentially create a separate 'offscreen' canvas that you can fill with some picture (in your case: set a single pixel). Then you swap the current canvas with the offscreen-canvas (the `SwapOnVSync()` conveniently returns the other canvas).
In your case, you would then want to just wait and swap that canvas back.

```
   offscreen_canvas->SetPixel(...)
   offscreen_canvas = SwapOnVSync(offscreen_canvas);
   sleep(exposure_time);
   offscreen_canvas = SwapOnVSync(offscreen_canvas);
```

tomasaidukas, 2018-07-30:
I tried your method to minimize the flickering, but the total power of the LED dropped significantly compared to when I just light it up and leave it. I saw a drop of about 100 times! Unfortunately this is too big of a problem for my application and I had to resort back to the 32x32 array I was using. Thanks for trying to help me though!

Also it might be a long shot, but if speed is an issue for controlling the 64x64 adafruit array (using my method, do you have any idea what I could do to optimize it? Is it mainly the digitalWrite() function for GPIO control (from wiringPI) that is the bottleneck? (I'm not a c++ coder as you might guess)

`
#include <wiringPi.h>
#include <stdio.h>
#include <stdlib.h>

void dec_to_bin(int dec, int* buffer){
// convert our integer into a 5 bit binary number
	int c, k, idx;
	
	idx = 0;
	for (c=4; c>=0; c--)	{
		// k starts with the MSB and keeps going towards LSB
    		k = dec >> c;  // shift dec values to the right by amount "c"
    		if (k & 1){
			buffer[idx] = 1;
    		}else{
			buffer[idx] = 0;
		}
		idx++;
	}
}

void led_x(int* binary_number){
	// binary_number says which LED must be turned on 
	// one particular ROW. There are two halves of the LED array
	// each containing 16 rows, hence a 4 bit number (0 to 15) 
	// specifies the row value for one particular half.
        digitalWrite(21,LOW);   // latch
        digitalWrite(4,HIGH);   // OE
        
        digitalWrite(22,LOW); // A0
        digitalWrite(26,LOW); // A1
        digitalWrite(27,LOW); // A2
        digitalWrite(20,LOW); // A3
        digitalWrite(24,LOW); // A4

        digitalWrite(22, binary_number[4]); // A0 -- LSB, pin A
        digitalWrite(26, binary_number[3]); // A1, pin B
        digitalWrite(27, binary_number[2]); // A2, pin C
        digitalWrite(20, binary_number[1]); // A3, pin D
        digitalWrite(24, binary_number[0]); // A4 -- MSB, pin E

}

void led_y(int y, int half, int color){
	// loop through each column of the row being illuminated
	// light up LEDs only for that column and color

        for (int j=1; j<=64; j++){
        	digitalWrite(17, LOW); // clk
        	digitalWrite(5, LOW); // R1
        	digitalWrite(13, LOW); // G1
        	digitalWrite(6, LOW); // B1
        	digitalWrite(12, LOW); // R2
        	digitalWrite(16, LOW); // G2
        	digitalWrite(23, LOW); // B2

		// when we hit our required column that we want to turn on.
		if (j == y){
			if (color == 0){
				if (half == 1){
		        		digitalWrite(5, HIGH); // R1
        				digitalWrite(13, LOW); // G1
        				digitalWrite(6, LOW); // B1
        				digitalWrite(12, LOW); // R2
        				digitalWrite(16, LOW); // G2
        				digitalWrite(23, LOW); // B2

				} else {
		        		digitalWrite(12, HIGH); // R2
        				digitalWrite(5, LOW); // R1
        				digitalWrite(13, LOW); // G1
        				digitalWrite(6, LOW); // B1
        				digitalWrite(16, LOW); // G2
        				digitalWrite(23, LOW); // B2
				}
			} else if (color == 1){
				if (half == 1){
		        		digitalWrite(13, HIGH); // G1
        				digitalWrite(5, LOW); // R1
        				digitalWrite(6, LOW); // B1
        				digitalWrite(12, LOW); // R2
        				digitalWrite(16, LOW); // G2
        				digitalWrite(23, LOW); // B2
				} else {
		        		digitalWrite(16, HIGH); // G2
        				digitalWrite(5, LOW); // R1
        				digitalWrite(13, LOW); // G1
        				digitalWrite(6, LOW); // B1
        				digitalWrite(12, LOW); // R2
        				digitalWrite(23, LOW); // B2
				}
			} else if (color == 2){
				if (half == 1){
		        		digitalWrite(6, HIGH); // B1
        				digitalWrite(5, LOW); // R1
        				digitalWrite(13, LOW); // G1
        				digitalWrite(12, LOW); // R2
        				digitalWrite(16, LOW); // G2
        				digitalWrite(23, LOW); // B2
				} else {
		        		digitalWrite(23, HIGH); // B2
        				digitalWrite(5, LOW); // R1
        				digitalWrite(13, LOW); // G1
        				digitalWrite(6, LOW); // B1
        				digitalWrite(12, LOW); // R2
        				digitalWrite(16, LOW); // G2
				}
			}
		}
       		digitalWrite(17,HIGH); // CLK
        }
        digitalWrite(21, HIGH); // yellow - Latch
        digitalWrite(4, LOW); // OE
}

int main (int argc, char** argv){
  /*
  Inputs x and y range from 1 to 32, color = 0,1,2
  */
  wiringPiSetupGpio  ();
  pinMode (5, OUTPUT) ; // R1
  pinMode (13, OUTPUT) ; // G1
  pinMode (6, OUTPUT) ; // B1
  pinMode (12, OUTPUT) ; // R2
  pinMode (16, OUTPUT) ; // G2
  pinMode (23, OUTPUT) ; // B2
  pinMode (17, OUTPUT) ; // CLK
  pinMode (4, OUTPUT) ; // OE
  pinMode (21, OUTPUT) ;// LATCH
  pinMode (22, OUTPUT) ;// A0
  pinMode (26, OUTPUT) ;// A1
  pinMode (27, OUTPUT) ;// A2
  pinMode (20, OUTPUT) ;// A3
  pinMode (24, OUTPUT) ;// A4 -- ADDED THE E PIN

  if (argc != 4){
	printf("Enter x,y coordinates and LED color \n");
	return 1;}

  int x = atoi(argv[1]);
  int y = atoi(argv[2]);
  int color = atoi(argv[3]);
  int half, x_half; // half defines which half of the LED array is used and x_half defines which row is used (from 1 to 32)

  if (x > 32){
	x_half = x - 33;
	half = 2;
  }else{
	x_half = x - 1;
	half = 1;
  }

  int binary_number[5]; // buffer to store the 5bit number which is sent through the ABCDE lines
  dec_to_bin(x_half, binary_number);

  // light up the LED
  led_x(binary_number);
  led_y(y, half, color);

  return 0 ;
}
`

hzeller, 2018-07-30:
I don't know what you measured when you got your 1/100 of brigthness. Did you set the PWM bits to 1 and set the full brightness of the pixel with 255,255,255 ?

w.r.t. the code:
I don't understand what you are doing with the `dec_to_bin()` function. The value is already stored in binary in the integer, so converting it in a complicated way to an integer buffer with hard-to-understand code is not helpful.

Then you also set first all ABCDE to low and then set again to the value you want. This is an unnecessary slow way; no need to set to zero before.

So in led_x(), simply pass the row address and do this to set the binary values:
```
digitalWrite(22, row_address & 0x01); // A0 -- LSB, pin A
digitalWrite(26, row_address & 0x02); // A1, pin B
digitalWrite(27, row_address & 0x04); // A2, pin C
digitalWrite(20, row_address & 0x08); // A3, pin D
digitalWrite(24, row_address & 0x10); // A4 -- MSB, pin E
```

Also, the code is a little hard to read as you switch x and y coordinates. The X-coordinate is along the axis where you clock in numbers, the y-coordinate is the one that needs to distinguish between sub-panels.

In general, you make your life easier if you count the coordinates from zero than from one. Because one-based requires you do do all kinds of corrections (e.g. the `x - 1` or `x - 33` part) or funky loops such as `for (int j=1; j<=64; j++)` while it would be more readable with `for (int j=0; j < 64; j++){`

So I haven't looked in detail, but there might be some off-by-one issues in the code because of that.

`digitalWrite()` might be too slow if you wanted to actually want to do PWM, but since you are only doing a single setting, the slow writes will not necessarily bother you.

I noticed that you exit immediately after you set the LED. Of course, wiringPi will probably reset all kinds of GPIOs afterwards, so this is not what you want. Instead, you want to sleep for the while your exposure is and then set `~OE` high again.

```
....
  led_y(...);
  usleep(exposure_time_in_microseconds);
  digitalWrite(4, HIGH);

  return 0;
```
(make sure to include `<unistd.h>` for usleep).

visansi, 2018-11-16:
@tomasaidukas it was fantastic to see you are working on FPM reconstruction!  were you able to get your setup working? 

I have the same need to keep the LEDs lighted up continuously for a few seconds for bright and dark field illumination control with led matrix for microscopy imaging.

I bought a [unicorn HD HAT] (https://shop.pimoroni.com/products/unicorn-hat-hd) and its too dim to give any useful illumination from their SMD2121(??) pixels. It was also not giving a large enough angle for dark field illumination (I also tried their 8X8 ws2812 LED HAT).  @hzeller just as you warned I see one DED pixels, even when just playing around with indexing the LEDs on the unicorn HD HAT. I dont believe I would have left it on for more than 5-10s. 

So I got an adafruit 32X32 matrix just now and I am trying to get it going - with non-existent electronics background.  Thanks for posting your wiringpi code - I see that it uses the same hardware mapping as the "adafruit-hat". I have connected it with the Adafruit Matrix Bonnet - and also without the bonnet (using the bonned only to route the 5V supply) and there is no sign of life. 

I [disabled the snd_bcm2835](https://www.instructables.com/id/Disable-the-Built-in-Sound-Card-of-Raspberry-Pi/) and dtoverlay=w1-gpio in boot config.

I am not sure where to start debugging this issue. Could you guess any obvious gotchas with controlling this type of LED matrix? 

[image]

[image]

@hzeller thanks for your fantastic work on the LED matrix code base. Regarding your concern on having the LEDs on always on mode, is there any outdoor LED matrix you would recommend or think that would withstand 5-10 seconds of always on state without risk of going DED?