hzeller/rpi-rgb-led-matrix #672: Is it possible to run with sudo

aussieW, 2018-07-28:
I am attempting to integrate with the Mycroft Voice Assistant (https://mycroft.ai/) and to do so I cannot run using sudo. Is it possible to give another user appropriate permissions to run the matrix display? If so could you please explain how.
Thanks

hzeller, 2018-07-28:
You need root access to be able to open the GPIO ports, so you have to be root to do so. So for that you either need to start it with sudo or as root directly.
If you need to start it as other user who does not have sudo permissions, you could consider setuid root of course.

What I usually would do however to bridge permission boundaries, is to start the led-matrix independently and receive images to display over the network. That way, you can totally separate the binaries and don't need to have one start the other (I usually use the [FlaschenTaschen protocol](https://github.com/hzeller/flaschen-taschen/blob/master/doc/protocols.md) for which there is a [network implementation for the rpi-rgb-led-matrix](https://github.com/hzeller/flaschen-taschen/tree/master/server#rgb-matrix-panel-display), but you can of course come up with your own).

aussieW, 2018-07-29:
Thanks for the quick response.

closed 2018-07-29 by aussieW