hzeller/rpi-rgb-led-matrix #673: 16x32 adafruit screens (daisychain) not working

tliu2019, 2018-07-29:
I followed the instructions on this github page to wire the display onto the pi, but for some reason I could not successfully run any of the demos on there. 

Only half of both screens light up a solid light blue color regardless of the demo that I am running. I am fairly confident that my wiring is correct. 
For example, I entered the code 
`sudo  ./demo -t 5 -D 1 runtext.ppm --led-no-hardware-pulse`
and produced results shown in the image below.

Any ideas on what went wrong? Is there something I might be missing in the command or code?
Thanks in advance!
-Tim
[image]

hzeller, 2018-08-02:
Did you follow all the instructions in the [troubleshooting section](https://github.com/hzeller/rpi-rgb-led-matrix#use-minimal-raspbian-distribution) ?
The fact that you use `--led-no-hardware-pulse` for instance tells me that you haven't disabled sound yet, probably other things as well not yet.

tliu2019, 2018-08-04:
Hi thanks for the reply,
I disabled the sound and added a few parameters in the command.
When I enter,
sudo  ./demo -D 1 runtext.ppm --led-rows=16 --led-cols=64 --led-brightness=40 --led-chain=2 --led-parallel=2

The screen works; however, when trying to run the python files, for some reason the screen still remains blank. Is there something I could be missing that's not in the instructions?

Command I used for the python file:
sudo python3 ./runtext.py