hzeller/rpi-rgb-led-matrix #677: Problem with pixel intensity in some rows

CamiloBravo, 2018-08-09:
Wonderful work with the library! But I have a problem likes shown below:

[image]

[image]

[image]

I have a Raspbian O.S.

Does anyone know why this happens?

These are the parameters that I used:

sudo ./demo --led-rows=16 --led-cols=32 --led-chain=1 --led-row-addr-type=2 --led-multiplexing=3 --led-brightness=20 --led-gpio-mapping=regular --led-pwm-bits=11 --led-scan-mode=0 --led-pwm-lsb-nanoseconds=500 -D 1 runtext16.ppm --led-slowdown-gpio=2

Thanks!

hzeller, 2018-08-09:
Interesting. I see you had to choose `--led-row-addr-type=2` for your panel to work; maybe there is some funky timing going on with your particular panel.
Can you take a picture of a grayscale/color image using the [`led-image-viewer`](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/utils#image-viewer), maybe this gives us more hints what is going on ?

CamiloBravo, 2018-08-09:
Hi Hzeller, thanks for reply!
With images worked perfect! Now my question is: how do I print images with a height greater than 48 pixels?

havituloh, 2025-01-20:
[image]
Hi Mr hzeller
why is there this problem? witing has followed what is in the readme, but it is still like this, the command that is run is like the following " sudo ./demo -D 0 runningtext.ppm --led-cols=64 --led-gpoo-mapping =regular --led-multiplexing=1 --led-slowdown-gpio=4 --led-chain=2 Sorry, my English is not good Thank You

davemaster, 2025-01-20:
You forgot to share your panel type/model, take pictures of your panel back, the a video and the commando of the result in THAT VIDEO... for a better diagnosing

havituloh, 2025-01-20:
My panel P5 hub75 and raspberry pi4

https://github.com/user-attachments/assets/d185ec49-92ee-4760-a296-741102cc75a0

[image]

[image]

davemaster, 2025-01-20:
Your panel, is 32x64 ??? 

No, it's 64x32, two modules chained

Now, ALWAYS, working with these panels, You must ensure you have the RIGHT POWER SUPPLY, you have 2 x 32 x 64 leds, Your power supply MUST BE at least, 5V 10A->15A

[image]

Also, search a connector with two (no one), TWO EACH positive and negative power connector.

ALSO, You must be sure, You have connected (or check the wires to those Pins A, B, C, _**D, and E..**_.. ). Even, You 're using a x32 rows panels, internally, the manufacturer ( this is the most possible) made some kind of arrangement for drive those panels... what can I possible know....

Once you have properly energized your panel, and those wires, and the pins A, B, C, and D, E, try with multiple led-multiplexing values, EACH VALUE, means a different kind of RGB PANEL... led-multiplexing=2, then 3, etc.

Share video, image, and command.... best regards

havituloh, 2025-01-22:
I've tried it but it doesn't work. Is it possible that the pin on the LED doesn't have E but ground? Is there a solution for this, Mr Hazller?
Thank's

https://github.com/user-attachments/assets/e988a5dc-5855-49f1-a990-57509ee35133
[image]

[image]

hzeller, 2025-01-22:
Please don't re-use old bugs (different brightness) with totally different issues (off LEDs).

In general, I don't suggest to directly wire the Pi to the panels, you can strain the GPIOs and possibly destroy them. So first I'd [build or buy](https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter) a level converter.

Your symptoms look a little bit like one of the address lines not working, but then it would also more look like an overlay of the stripes.

davemaster, 2025-01-22:
> Please Mr davemaster help me

change A, B, C, D wires, or re-test continuity in each of them.. then, TRY AGAIN with only ONE PANEL... the text ppm demo, ONLY ONE panel... post video or image of output

closed 2025-11-27 by marcmerlin