hzeller/rpi-rgb-led-matrix #678: Text via Command

ace16vitamine, 2018-08-11:
Hi,

how can I display a text string without scrolling to my display?

I know that  I can work with stdin:

./text-example -f ../fonts/7x13.bdf --led-rows 16 --led-gpio-mapping=adafruit-hat --led-cols 64 -y 5 -C 255,0,0

But how can I display the text directly submitted via command? My Idea was:

./text-example -f ../fonts/7x13.bdf --led-rows 16 --led-gpio-mapping=adafruit-hat --led-cols 64 -y 5 -C 255,0,0 **| echo "TEST"**

but this does not work. Any other Ideas?

Thanks

hzeller, 2018-08-11:
Pipes work from left to right in the shell just like the direction of reading. What you output in the first command, you pipe into the next.
So you need to do something like
```
  echo "TEST" | ./text-example ...
```

ace16vitamine, 2018-08-11:
Hi, absolutely, you are right! But if I Try 

`echo "TEST" | ./text-example -f ../fonts/7x13.bdf --led-rows 16 --led-gpio-mapping=adafruit-hat --led-cols 64 -y 5 -C 255,0,0`

I did not receive any output on the matrix.

ace16vitamine, 2018-08-11:
I found a solution to refresh:
`(while :; do cat Anzeige.txt ; sleep 0.2 ; done) | sudo ./text-example -f ../fonts/7x13.bdf --led-rows 16 --led-gpio-mapping=adafruit-hat --led-cols 64 -y 1 -C 255,0,0`

Thanks

closed 2018-08-11 by ace16vitamine