hzeller/rpi-rgb-led-matrix #679: Fill() not respecting RGB sequence

mikepparks, 2018-08-12:
I noticed when testing my [64x64 panel](https://www.adafruit.com/product/3649) using the newer [Adafruit HAT](https://www.adafruit.com/product/3211) that it seems the framebuffer Fill() method is not respecting the RGB sequence.  I switched to "RBG" because it seemed that SetPixel wasn't working as expected in "RGB" mode, but noticed that doing this did not affect the output of Fill for the B & G colors at all. Is this happening to anyone else, or is this the expected behavior? I'm currently poking around for a fix, but would love to make sure that I'm not chasing windmills first in case there's something I missed.

hzeller, 2018-08-12:
So it works for SetPixel(), but not Fill() ?

mikepparks, 2018-08-12:
Correct.

flavio-fernandes, 2018-09-19:
@hzeller sorry to bother you, but on my setup, the commit ba3226b873dcef93f8ff64a1836023d9d4b6a844
really broke python binding.

To reproduce, simply use:
```
options.led_rgb_sequence = "RBG"
```
Let me know if you need more details, but it seems that seq is pointing to the wrong string,
or to a string that has been freed up.

```
#  Framebuffer::GetGpioFromLedSequence  is very unhappy
$ sudo ./screen.py
LED sequence '
�t' does not contain any 'R'.
Aborted
```

hzeller, 2018-09-19:
Interestng, did it even work before then @flavio-fernandes ? Because the lifetime assumptions about this string didn't change at all.

You haven't showed your code. Are you by any chance using cppinc.Options class in Python ? That won't work as it does not copy the string locally, while RGBMatrixOptions does.

flavio-fernandes, 2018-09-19:
heh, sorry for the confusion. I will make a PR to give you an example on how it breaks.
I have actually made no changes at all to this repo. Simply using Adafruit's install script and hacking the commit it pulls from your repo [1].
I only see this issue if I use commit ba3226b.

[1]: https://github.com/flavio-fernandes/Raspberry-Pi-Installer-Scripts/commit/96a8219c6736bfa3dd650aa793cb4106314e0adf#diff-bb0ce91e3af2d6fc15fd58897ee96e47R10

hzeller, 2018-09-20:
For a baseline test, I compiled the library and did a fresh compilation and install of the python binding and the following works on my Pi with Python 2.7
```
  sudo bindings/python/samples/rotating-block-generator.py --led-rgb-sequence=BRG
```

What is different in your set-up ?

hzeller, 2018-09-20:
Ok, so looking at your code
  https://github.com/flavio-fernandes/bedclock/blob/master/bedclock/screen.py#L70
you create the options struct on the stack, so the string content would actually be destroyed; then if you later create a frame, it will choke there.

Fixed. With the latest commit, it should now work.