hzeller/rpi-rgb-led-matrix #680: Execute without sudo

Baptiste-Garcin, 2018-08-13:
Hi,

Thanks for this awsome library.

I would like to know if there is a way to run binary without sudo-ing ?

JokerCatz, 2018-12-01:
I tried use C lang create a unix socket server and use Golang send package to C , it only C part need root , and it worked : )

JokerCatz, 2018-12-01:
I upload my C lang server and Golang client here : )

https://gist.github.com/JokerCatz/b39143861480ef174069bc5c718b142c
https://gist.github.com/JokerCatz/9790290776b71f7cff41f7bd060cd83b

boilerbots, 2019-06-09:
There is a solution, I just implemented it.

First open /dev/gpiomem instead of /dev/mem, now you can run without root privileges.

Second thing that requires root is changing the priority level of the scheduler. For this you can simply do `sudo setcap 'cap_sys_nice=eip' $@`  in the makefile after the final linking.

hzeller, 2019-06-09:
> There is a solution, I just implemented it.
>
> First open /dev/gpiomem instead of /dev/mem, now you can run without root
> privileges.

Does it work also with the other parts needed to do hardware pulse output ?

> Second thing that requires root is changing the priority level of the
scheduler. For this you can simply do sudo setcap 'cap_sys_nice=eip' $@ in
the makefile after the final linking.

boilerbots, 2019-06-09:
> Does it work also with the other parts needed to do hardware pulse output ?
> Second thing that requires root is changing the priority level of the
> scheduler. For this you can simply do sudo setcap 'cap_sys_nice=eip' $@ in the makefile after the final linking.
> […](#)

Good question, I am running with --led-no-hardware-pulse because I am trying to preserve the audio. I suspect it won't but I haven't looked into that.

commit 7fdbf1ed (2019-07-06): Update Pi Model detection; allow degraded non-root operation.

o Update PI model detection. Now detects Raspberry Pi 4 (fixes #840)
  and Rasberry Pi 3 A+ (fixes #727)
o If we can't mmap /dev/mem, fall back to /dev/gpiomem. Unfortunately,
  this will not allow to read the 1Mhz timer nor allow for PWM to work
  properly (fixes #680)
closed 2019-07-06 by hzeller