hzeller/rpi-rgb-led-matrix #683: Flickering on the leading edge

hemantjha28, 2018-08-16:
Hi,

I have to create a p3 [64 X 64] screen with 12 modules [3 parallels with 4 modules in chain]

As a test case I created a chain  of 2 p3 modules [64 X 64] and following is the result that I am getting.

I am not sure what will be the right technical term to describe this problem but as you can see it is not clear with a lot of pixels blinking on the leading edges ahead of the characters.

Please help me with a solution

I am using a Raspberry pi zero W with a 5 V 40 amps power supply. I am not using the active adapter pcb, i have simply hand soldered a prototyping board connecting the pins as suggested by you.

Is it a problem because I am not using active adapter pcb?

Also PCB designers in India are taking some time to get this printed. This there a PDF version of the PCB Design so that I can try etching the PCB myself at home.

Please find the attached images and videos.

Sincere Thanks Again,

Hemant
[image]
[VID_20180816_104952078.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/2292520/VID_20180816_104952078.zip)

hzeller, 2018-08-16:
The ghosting might might be issues with detecting the voltage level, so maybe sometimes the edge of the clock signal is not detected well.
So if you have the active digital level converter 3.3V -> 5V, that might help.
You can try using a different gpio slowdown value.

Do not use a Pi Zero. It is has a very slow, single-core CPU and you will not get enough refresh rate in particular with Get a Pi 3.

hemantjha28, 2018-08-16:
Hi Thanks Again.

closed 2018-09-20 by hzeller