hzeller/rpi-rgb-led-matrix #685: Adafruit Matrix Hat Bonnet with HUB-75E 64x64 Matrix

DonDalle, 2018-08-28:
Hello There,

I have some troubles getting my HUB-75E Matrix to work with the Adafruit Hat Bonnet. I did shorten the circuit of the E-Line on the Bonnet as they described on their website. I provide a 5V 10A power supply to the matrix and I'm running the program with following command:

`sudo ./demo -D8 --led-rows=64 --led-cols=64 --led-chain=1  --led-gpio-mapping=adafruit-hat 
`

The Matrix just stays blank. I don't get any output on it.  I did go through the troubleshooting section and it didn't solve my problem. I noticed a weird behavior: When I disconnect the Bonnet from the PI and connect it again while both the Pi and Matrix are on, I get some weird output on the panel and it does change when I start and stop the program. 

I also have a 64x32 Matrix and it works like a charm with the very same setup. 

This is the pinout of the Matrix: 
[image]

[This is the eBay link to the Matrix I use ](https://www.ebay.de/itm/4x-P2-5-ICN2053-high-quality-LED-Matrix-Panels-Display-Austauschmodule-64x64-/302838587824?_trksid=p2385738.m4383.l4275.c10)

Do you have any Ideas how to troubleshoot this? 

Thank you very much!

hzeller, 2018-08-28:
What description did you follow for the E-line ?
But even if there is no E line, you should get at least some partial output.

I could only imagine that due to the E-line (which usually would be GND), that GND might not be properly connected, but I don't know the Bonnet hat. Try to make sure to connect the GND of the Pi with the GND of the matrix.

DonDalle, 2018-08-29:
Thanks. The Bonnet is basically like the normal Adafruit Hat but comes assembled. The mod I did was to solder two pins on the back so that one of the ground lanes becomes the E-Line.

However, I did connect the Matrix directly to the Pi now without the Hat. And it doesn't light up too. I double checked the wiring and did connect every cable correctly. 

I did try something though: I connected my old 64x32 Panel to the 64x64 in Daisy Chain like this:

Pi > [64x64] > [64x32]

Now the 64x32 Panel did light up while the 64x64 which is first in line didn't light up. I did run the color cycle and it showed all colors but glitched.  Both are connected to the same power supply. When I disconnect the power from the 64x64, the signal isn't transferred to the second one. This tells me that power isn't the issue here. 

I didn't try the vice versa way yet but will eventually do it. I also tried different 64x64 Panels and it din't work with them either. 

Do you know what the Problem could be here? 

Thanks again!

foliastar, 2018-09-30:
Dear all,

I have the same LED matrix but it stay blank...did some one got them working?
At the moment I am notz sure if I did all correct on my PI ;-))

I installes the Rasbian pi and copied all data to the root folder......could this work?

Sascha

Cellgalvano, 2018-09-30:
Hi Everyone!

I also got four of these Panels (from the same eBay seller). I tried to write a simple Arduino sketch to get at least something, but I failed. I can't get them to light up. 

Am I right with the assumption that these Panels should be a lot better and faster?
Instead of constantly updating the frame we should only update the integrated framebuffer when there is a new image to display. Does anybody know how the scan line switching works? As far as I read through the various datasheets the next scan line should be selected after 1024 GCLK pulses. I measures the pins on the HUB-75E connector and found that the CLK goes to DCLK, LATCH goes to LE and OE goes to GCLK. 

Felix

hzeller, 2018-09-30:
You should probably concentrate on #466 which is focusing on figuring out these kind of panels @Cellgalvano 

I am going to close this one, so that it is not distracting from that.

closed 2018-09-30 by hzeller