hzeller/rpi-rgb-led-matrix #686: Python Binding Image Samples not clear

BryceARich, 2018-08-29:
Hello,

I've been introducing myself to the library as well as doing my first project with a Raspberry Pi 3 (model B) running the base rasbian OS. I was able to successfully run the demos with the static images I hope to use in the `examples-api-use/demo` and get clear images. When trying to display the same images using `image-viewer.py` in python3 I am getting smeared images. The images are unclear, there appears to be extra noise outside of the range of the image, and the image is inconsistent (flickering).

At the moment I am waiting on parts (74HCT245) as suggested in the Troubleshooting section, but have been using `--led-slowdown-gpio=2` for both the demo and python samples. Are there other possible issues I'm not accounting for on either the software or hardware side that could be causing this difference between the c++ and python samples? I have not yet edited any of the code as I wanted to ensure I had all of my settings correct first.

I appreciate any guidance that could be given and am really excited to use this library.

ngrices, 2018-09-05:
@BryceARich I cant get images to display on my 64x64 LCD at all. Are you passing in jpg or png images to PIL? or are you pre-converting images to PPM?

lnestoros, 2018-09-07:
Is it just images, or is it all of the examples in the sample section?  i am experiencing blurring and flickering with every python sample.  Is this common?

https://drive.google.com/open?id=1xW4ab0VBtOQd1h-krNYyv0Q9_YJePZi0
https://drive.google.com/open?id=1u1omO3EQvgBHW-qZxscLfRmioOZhC3ii

When i went through the "C" portion of the readme, i was able to successfully reduce the flicker by utilizing the options for led-slowdown-gpio. led-pwm-lsb-nanoseconds, and pwm-dither-bits.  I do not see an option in the python binding for pwm-dither-bits so i am at a bit of a loss for how to help further.

hzeller, 2018-09-07:
Did you make sure to use the isolcpus setting ?

The flickering seen in your video do look like mostly electrical issues (too long cables, no 3.3V->5V conversion), so you want to address these first. And of course stable power supply; if you have too much noise on the power supply, add large, low-ESR capacitors to the power inputs etc.

Other than that, you can try to use a higher level of slowdown GPIO to see if slower clock-speed can tame the matrix.

Python will in general generate more flicker as it is a language with random garbage collection times, which will tax the memory bus that is limited on the Pi. But it does not seem to be a primary problem in this particular case.

closed 2025-11-27 by marcmerlin