hzeller/rpi-rgb-led-matrix #689: Completed Scoreboard, used at a game

Beersatron, 2018-09-11:
I finally got my scoreboard to the point that it would work at the field for a game. I took it to our last home game of the year but only had an old UPS which said it would power it for 156 minutes but lasted 20 minutes - bit of a disappointment.

We had an away game in Baton Rouge last Saturday which is 4.5 hours drive away. I worked late on Thursday night, tweaking the software, and then late Friday night, tweaking the frame. We sourced a generator for the field and to ensure good quality power I passed it through the UPS.

End result was a fully functional scoreboard that could be seen at least 60ft away in the very hot, very humid, very bright Louisiana day.

[image]

The refresh rate was around 120-140Hz, which meant that none of our smart phones could get a good picture.

[image]

But [this picture](https://www.instagram.com/p/BnmavsIgBGX/?taken-by=houston_lonestars) was taken by somebody with a digital SLR who lowered the shutter speed. It is at the end of the 1st Quarter, we ended up winning handily by the end of all 4 quarters.

mbates14, 2018-09-13:
Curious how you wired that. There are only 3 ports on the regular buffer card.

Beersatron, 2018-09-13:
I used 2 parallel chains. I don't have a better picture of the back than this, for some reason.

[image]

The Wattage draw when on and running is ~34w

[image]

The UPS that I was using was old and catastrophically failed at around 20mins, no matter what load I put on it.

hzeller, 2018-09-28:
Very nice project, congratulations!

mbates14, 2018-10-03:
> I used 2 parallel chains. I don't have a better picture of the back than this, for some reason.
> 
> [image]
> 
> The Wattage draw when on and running is ~34w
> 
> [image]
> 
> The UPS that I was using was old and catastrophically failed at around 20mins, no matter what load I put on it.

Its probably the battery in the UPS, Time to changer her out and itll be good as new!

mbates14, 2018-10-31:
Figured I would post what I have done with it so far. 

https://drive.google.com/open?id=173Unv9Tm1T2UpT4iKOwIBZ7nyVy3Rw7Q

Camera did a horrible job picking up the images because of the refresh, but youll get the idea.

It's a Team Name Caption panel I designed that goes into a scoreboard.

jtaylor54, 2018-11-12:
Would you be interested in sharing your code? I am especially interested in the web interface you used to control it. Maybe some pointers anyway. 
Tx
JT

closed 2025-11-27 by marcmerlin