hzeller/rpi-rgb-led-matrix #690: Condensed Fonts?

mbates14, 2018-09-13:
Does anyone have any good condensed monospace fonts? 

I am trying to fit 8 characters within a 96x32 array which is 6 16x32 tiles. Trying to find a font that can fit the entire height of array as well. 

So a tall narrow monospaced font. Any thoughts on this? 

I been fontforging all day long and getting nowhere fast.

KentWalker, 2018-09-13:
You can alway create your own BDF font to suit.  It's not terribly hard, but it is very time consuming since you will need to define every character that you intend to use.  The plus side is that you have full control over how each character will be displayed.  See [https://en.wikipedia.org/wiki/Glyph_Bitmap_Distribution_Format](url)

There is a BDF editor called xmbdfed but I don't think there is a Debian distribution so you may not get it to run on Ubuntu or Raspberry Pi.  BDF is quite old now and not very well supported. See [https://www.mankier.com/1/xmbdfed.man#](url)

You might be able to do a manual install of xmbdfed - [https://fossies.org/linux/misc/old/xmbdfed-4.7.tar.gz/](url)  but no guarantees that it will work or even compile, it dates back to 2004.

Another editor is [https://github.com/andrewshadura/gbdfed](url) - again it will require manual compilation.

Then of course you could just open an existing file (say 9x15.bdf) in a text editor and hack it to suit.  If you want 8 characters to fill 96x32 pixels I would suggest a 12x32 font.  If you stick to only the 7 bit ASCII characters (from ' ' to '~') then you can cut the file size down too.

Mayki07, 2018-09-16:
If anyone could make CE font 14x32 I would be very grateful to him. I would need it for my project.

hzeller, 2018-09-16:
Note, if you have a true type font file, you can use `otf2bdf` to convert them to a bdf file. This call would generate a 30 point bdf font for instance:
```
  otf2bdf -o myfont.bdf -p 30 /path/to/myfont.ttf
```

closed 2018-09-28 by hzeller