hzeller/rpi-rgb-led-matrix #695: python multiprocessing

dadrake3, 2018-09-28:
My goal is to have the parent process be a dispatch function that receives input from an ir remote, that starts a subprocess to do a certain led function, and the parent process can then either signal to that child process or kill it based on remote input. The problem I have though is when the child process reaches "offscreen_canvas = self.matrix.SwapOnVSync(offscreen_canvas)" it hangs and does nothing. 

my code is like this: 

p = Process(target=self.LED_loop, args=(context,))
p.start()
p.join()

I am able to spawn threads that handle LED functions but I need to use multiprocessing as I cant signal or kill threads

thanks

hzeller, 2018-09-28:
Depending if you already started the matrix when forking influences if things work; because if you fork, all threads are stopped and you end up with no update thread.
So you need to make sure to initialize the matrix after the fork.

However, I am not sure if it works to kill and restart, this really depends on how Python does its garbage collecting. These kinds of things are easier done in C++ because it is very clear what is happening.

Anyway, starting and killing the LED process will result in LED glitches, as the boards don't take it well if they are powered but don't get updates.

I'd suggest have the LED matrix run all the time and use sockets or pipes to communicate updates to it.

dadrake3, 2018-09-28:
so I was having the fork happen inside of the run function, could you explain what you mean exactly by initialize the matrix?

hzeller, 2018-09-28:
you have to initialize the matrix in your subprocess, and there do your updates.
But you can not fork inside your update process of course, as forking means, a complete copy of the process is created, with all threads stopped.

At best, you can create a new thread and do something there.

dadrake3, 2018-09-28:
The issue I have though is I need to be able to arbitrarily kill or signal the thread / process, and python threading does not support either. Its also harder to check if a thread is active. But i see what you mean that I cannot fork from within the update process. Could you explain more by what you mean sockets or pipes to communicate with it? Is there support for something like that in python?

Thanks

hzeller, 2018-09-28:
if you have threads, you can also use the usual techniques with Mutexes or condition variables. Sockets and Pipes are more useful if you have processes.
I am a C++ guy, so I don't really know much Python to suggest good ways to do it there.

closed 2018-09-28 by hzeller