hzeller/rpi-rgb-led-matrix #698: video-viewer playback issue

mbates14, 2018-09-29:
I had some small content made by another person for me to display on my 32x96 pixel matrix. 

Everything I have has played on it and converted to stream files without issues. Except for what he sent me. 

What he sent me will play back in this weird corrupted starburst color pattern. 

If I convert it to stream, sometimes it plays fine, sometimes it plays in that corrupted starburst pixel state. 

It also keeps popping up this: 
[swscaler @ 0x1f68580] No accelerated colorspace conversion found from yuv420p to rgb24.

I tried using ffmpeg in windows to convert the file format from mp4 to wmv, and to other formats and codecs. They all do that color corruption gig. 

I attached the file so maybe someone could tell me the exact cause? 
[homerun1.zip](https://github.com/hzeller/rpi-rgb-led-matrix/files/2430694/homerun1.zip)

Any thoughts?

hzeller, 2018-09-29:
Oh, I once had a bad no-name SD card that ruined my day like that. Since then, I only get genuine SanDisk cards, because a wasted afternoon is way more expensive and frustrating than the additional $/€/£ the good card costs.

Have fun with your now working display!

mbates14, 2018-09-29:
Sadly, this one was a Sandisk 32GB. 

But I think that "I" may be the cause, as I tried to back up the SD card in its current state before I continued on, and I used PiShrink with the image, and re-wrote the image to make sure everything was good. That seemed to be when all the problems started. So PiShrink is a strong possibility of the cause.  Anyways. 

Yea, Fun haha. I am using them for our Team Name Captions in our scoreboards. the RGB tiles are cheap from our supplier, and they are the same tiles used in the big video displays that we buy from overseas. So its a win-win situation.